adafruit/Adafruit_Blinka #690: board.I2C() AttributeError: 'module' object has no attribute 'I2C'

Miner34dev, 2023-07-20:
Hello, i'm trying to use a msa301 on a RPi Pico running Micropython.
Sadly, when i try to run the first example, it trows an error:
```python
Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
AttributeError: 'module' object has no attribute 'I2C'
```
I reinstalled all the libraries 3 times, but it still doesn't work :disappointed:.
There is the code i was trying to execute:
```python
import time
import board
import adafruit_msa3xx

i2c = board.I2C()
msa = adafruit_msa3xx.MSA301(i2c)
```
Can somebody help me?

tannewt, 2023-07-20:
Init the I2C with explicit pins using `busio.I2C()`. The pico doesn't have a default set of pins labeled for I2C.

closed 2023-07-20 by tannewt

Miner34dev, 2023-07-20:
Like this?
```python3
import time
import board
import busio
import adafruit_msa3xx

i2c = busio.I2C(0, 1)
msa = adafruit_msa3xx.MSA301(i2c)
```