adafruit/Adafruit_Blinka #692: Orange PI 5 PLUS RK3388 no support?

drozdi70, 2023-07-27:
Does Adafruit-Blinka support Orange Pi 5 Plus? 
(http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-5-plus.html)

if I query

orangepi@orangepi5plus:/usr/local/lib/python3.10/dist-packages/adafruit_platformdetect$ cat [board.py](http://board.py/) |grep orangepi5

        elif board_value == "orangepi5":

if I run blinka_[test.py:](http://test.py/)

orangepi@orangepi5plus:~/work$ sudo python3 blinka_[test.py](http://test.py/)
Traceback (most recent call last):
  File "/home/orangepi/work/blinka_[test.py](http://test.py/)", line 2, in <module>
    import board
  File "/usr/local/lib/python3.10/dist-packages/[board.py](http://board.py/)", line 354, in <module>
    raise NotImplementedError(
NotImplementedError:
        Adafruit-PlatformDetect version 3.47.0 was unable to identify the board and/or
        microcontroller running the Linux platform. Please be sure you
        have the latest packages running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

How I can I fix it? :)

drozdi70, 2023-07-27:
I attach pinout made based on OrangePi_5_Plus_RK3588_user manual_v1.5.1.pdf (point 3.18.5.)

[orangepi5plus.zip](https://github.com/adafruit/Adafruit_Blinka/files/12187300/orangepi5plus.zip)

makermelissa, 2023-09-01:
Duplicate of #666

drozdi70, 2023-09-02:
This is a different board :(Gpio Orange pi 5 and Orange pi 5 plus differs...Temat: Re: [adafruit/Adafruit_Blinka] Orange PI 5 PLUS RK3388 no support? (Issue #692)Data: 2023-09-02 0:33Nadawca: "Melissa LeBlanc-Williams" ***@***.***&gt;Adresat: "adafruit/Adafruit_Blinka" ***@***.***&gt;; DW: "DroZDi" ***@***.***&gt;; "Author" ***@***.***&gt;; 

Duplicate of #666

makermelissa, 2023-09-02:
Oops, you're right.

drozdi70, 2023-09-03:
HiIf you need me to test im ready :)KindlyTemat: Re: [adafruit/Adafruit_Blinka] Orange PI 5 PLUS RK3388 no support? (Issue #692)Data: 2023-09-02 22:33Nadawca: "Melissa LeBlanc-Williams" ***@***.***&gt;Adresat: "adafruit/Adafruit_Blinka" ***@***.***&gt;; DW: "DroZDi" ***@***.***&gt;; "Author" ***@***.***&gt;; 

Oops, you're right.

drozdi70, 2023-11-25:
Probably you know the issue is related to the fact that both boards orange pi 5 and orange pi 5 plus are identified with the same  board_id ORANGE_PI_5 under adafruit package. Hardware is different and no way to work on both boards like that in the same time. My fix to work on orange pi 5 boards is to replace 2 files:

sudo cp -f orangepi5plus.py  /usr/local/lib/python3.11/dist-packages/adafruit_blinka/board/orangepi/orangepi5.py
sudo cp -p pin5plus.py /usr/local/lib/python3.11/dist-packages/adafruit_blinka/microcontroller/rockchip/rk3588/pin.py

Limits:
It will open only rk3588-i2c2-m0 rk3588-uart6-m1 rk3588-uart4-m0 on OPI5Plus.

[adafruit.zip](https://github.com/adafruit/Adafruit_Blinka/files/13465298/adafruit.zip)

makermelissa, 2023-11-29:
Fixed by #742

closed 2023-11-29 by makermelissa

drozdi70, 2023-11-29:
HiAre you sure it is fixed? :)As I put in https://github.com/adafruit/Adafruit_Blinka/issues/692we need also change somehow pin.py file in /usr/local/lib/python3.11/dist-packages/adafruit_blinka/microcontroller/rockchip/rk3588/ as it is shared with orange pi 5....orangepi5plus.py is not enough....Best regards/73RobertTemat: Re: [adafruit/Adafruit_Blinka] Orange PI 5 PLUS RK3388 no support? (Issue #692)Data: 2023-11-29 20:49Nadawca: "Melissa LeBlanc-Williams" ***@***.***&gt;Adresat: "adafruit/Adafruit_Blinka" ***@***.***&gt;; DW: "DroZDi" ***@***.***&gt;; "Author" ***@***.***&gt;; 

Closed #692 as completed.

makermelissa, 2023-11-30:
Ok, I'm going to reopen this, but rel-abel it as a bug.

makermelissa, 2023-11-30:
Orange Pi 5 PLUS support was added in #742. Based on comments in that PR, it sounds like the following need to 
happen:
1. Copy RK3588 microcontroller to RK3588S
2. Update any boards that are actually RK3588S to use the RK3588S microcontroller pin file
3. Make Changes to the RK3588 microcontroller pin file so that it is accurate
4. Orange Pi 5 should use the update RK3588 microcontroller pin file.

Note that I have neither an Orange Pi 5, nor an Orange Pi 5 PLUS, so these changes will probably be based entirely on data sheets and other info available online.

drozdi70, 2023-11-30:
I have both boards so I can make tests :)Just give me the input data :)RegardsRobertTemat: Re: [adafruit/Adafruit_Blinka] Orange PI 5 PLUS incorrect RK3388 definition (Issue #692)Data: 2023-11-30 17:24Nadawca: "Melissa LeBlanc-Williams" ***@***.***&gt;Adresat: "adafruit/Adafruit_Blinka" ***@***.***&gt;; DW: "DroZDi" ***@***.***&gt;; "Author" ***@***.***&gt;; 

Orange Pi 5 PLUS support was added in #742. Based on comments in that PR, it sounds like the following need to
happen:

Copy RK3588 microcontroller to RK3588S
Update any boards that are actually RK3588S to use the RK3588S microcontroller pin file
Make Changes to the RK3588 microcontroller pin file so that it is accurate
Orange Pi 5 should use the update RK3588 microcontroller pin file.

Note that I have neither an Orange Pi 5, nor an Orange Pi 5 PLUS, so these changes will probably be based entirely on data sheets and other info available online.

makermelissa, 2023-11-30:
Sounds good. Thanks.

makermelissa, 2023-12-04:
Looks like I forgot to actually re-open this.

drozdi70, 2023-12-10:
HiPlease tell me if adafruit is also certified to worki on Debian 12/13 and Ubuntu?KindlyRobertTemat: Re: [adafruit/Adafruit_Blinka] Orange PI 5 PLUS incorrect RK3388 definition (Issue #692)Data: 2023-11-30 17:42Nadawca: "Melissa LeBlanc-Williams" ***@***.***&gt;Adresat: "adafruit/Adafruit_Blinka" ***@***.***&gt;; DW: "DroZDi" ***@***.***&gt;; "Author" ***@***.***&gt;; 

Sounds good. Thanks.

drozdi70, 2023-12-17:
after last changes in Adafruit package I got:
orangepi@orangepi5-plus:~/work/openplotter_opi5plus$ python3
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.11/dist-packages/board.py", line 143, in <module>
    from adafruit_blinka.board.orangepi.orangepi5plus import *
  File "/usr/local/lib/python3.11/dist-packages/adafruit_blinka/board/orangepi/orangepi5plus.py", line 6, in <module>
    from adafruit_blinka.microcontroller.rockchip.rk3588 import pin
  File "/usr/local/lib/python3.11/dist-packages/adafruit_blinka/microcontroller/rockchip/rk3588/pin.py", line 11, in <module>
    GPIO0_A0 = Pin((0, 0))
               ^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 36, in __init__
    self._chip = gpiod.Chip(str(pin_id[0]), gpiod.Chip.OPEN_BY_NUMBER)
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: type object 'Chip' has no attribute 'OPEN_BY_NUMBER'
>>>

drozdi70, 2023-12-18:
Problem solved nad fixed

moursbourn1, 2023-12-21:
This still seems to be broken.   I have an Orangepi5 plus.  I am on /dev/i2c-2 (I2C2_M0) with:
3.3v Pin 1
SDA pin 3
SCL pin 5
GND pin 6

I can see the oled screen address:
```
orangepi5-plus:~$ sudo i2cdetect -y 2

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
detect finds the board:  Orange Pi detected.

but blinkatest gives me this:
```
Hello blinka!
Traceback (most recent call last):
  File "/home/hmc/Code/blinka_test.py", line 12, in <module>
    i2c = busio.I2C(board.SCL, board.SDA)
  File "/home/hmc/Code/.venv/lib/python3.10/site-packages/busio.py", line 37, in __init__
    self.init(scl, sda, frequency)
  File "/home/hmc/Code/.venv/lib/python3.10/site-packages/busio.py", line 162, in init
    raise ValueError(
ValueError: No Hardware I2C on (scl,sda)=((0, 15), (0, 16))
Valid I2C ports: ((0, (4, 21), (4, 22)), (1, (0, 13), (0, 14)), (1, (1, 9), (1, 10)), (3, (3, 15), (3, 16)), (5, (1, 14), (1, 15)), (7, (4, 10), (4, 11)), (8, (3, 18), (3, 19)))
```

drozdi70, 2023-12-22:
add to blinka_test.py:

print("Board id is: ",board_id)

and provide the output.

and next try this (please make copy of files: pin.py, orangepi5.py, orangei5plus.py before you start):

if you see ORANGE_PI_5 you need to 

sudo cp -f orangepi5.py  /home/hmc/Code/.venv/lib/python3.10/site-packages/adafruit_blinka/board/orangepi/orangepi5plus.py
sudo cp -p pin5plus.py /home/hmc/Code/.venv/lib/python3.10/site-packages/adafruit_blinka/microcontroller/rockchip/rk3588/pin.py

else ORANGE_PI_5_PLUS

sudo cp -f orangepi5plus.py  /home/hmc/Code/.venv/lib/python3.10/site-packages/adafruit_blinka/board/orangepi/orangepi5plus.py
sudo cp -p pin5plus.py /home/hmc/Code/.venv/lib/python3.10/site-packages/adafruit_blinka/microcontroller/rockchip/rk3588/pin.py

moursbourn1, 2023-12-22:
board_id needs to be defined as something?  what is it testing?

JasonJasu, 2024-02-04:
Workaround provided [here](https://github.com/adafruit/Adafruit_Blinka/issues/692#issuecomment-1867514988) seems to be working on my side, hence it seems that RK3588 pin file has the wrong definition. This change needs to be merged soon

3diamond, 2024-03-08:
> add to blinka_test.py:
> 
> print("Board id is: ",board_id)
> 
> and provide the output.
> 
> and next try this (please make copy of files: pin.py, orangepi5.py, orangei5plus.py before you start):
> 
> if you see ORANGE_PI_5 you need to
> 
> sudo cp -f orangepi5.py /home/hmc/Code/.venv/lib/python3.10/site-packages/adafruit_blinka/board/orangepi/orangepi5.py
> 
> else ORANGE_PI_5_PLUS
> 
> sudo cp -f orangepi5plus.py /home/hmc/Code/.venv/lib/python3.10/site-packages/adafruit_blinka/board/orangepi/orangepi5plus.py sudo cp -p pin5plus.py /home/hmc/Code/.venv/lib/python3.10/site-packages/adafruit_blinka/microcontroller/rockchip/rk3588/pin.py
> 
> maybe it can help you :) files related you can find on my github repositories...

Hi
Thank you all for for your efforts. Tried to get my project to work for 2 months without disturbing you guys. I really need your expertise and help. I am new to adafruit and blinka. I tried to follow the method above but I can't seem to get it to work. 
I am on /dev/i2c-2 (I2C2_M0) with:
3.3v Pin 1
SDA pin 3
SCL pin 5
GND pin 6

Here are the results I get:
Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> import digitalio
>>> import busio
>>> 
>>> import sys
>>> 
>>> import adafruit_platformdetect.constants.boards as ap_board
>>> from adafruit_blinka.agnostic import board_id, detector
>>> 
>>> print("Hello blinka!")
Hello blinka!
>>> print("Chip id: ", detector.chip.id)
Chip id:  RK3588
>>> print("Board id is: ",board_id)
Board id is:  ORANGE_PI_5_PLUS
>>> i2c = busio.I2C(board.SCL, board.SDA)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.10/dist-packages/busio.py", line 37, in __init__
    self.init(scl, sda, frequency)
  File "/usr/local/lib/python3.10/dist-packages/busio.py", line 162, in init
    raise ValueError(
ValueError: No Hardware I2C on (scl,sda)=((0, 15), (0, 16))
Valid I2C ports: ((0, (4, 21), (4, 22)), (1, (0, 13), (0, 14)), (1, (1, 9), (1, 10)), (3, (3, 15), (3, 16)), (5, (1, 14), (1, 15)), (7, (4, 10), (4, 11)), (8, (3, 18), (3, 19)))
>>> i2c = busio.I2C(board.I2C2_SCL_M0, board.I2C2_SDA_M0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'board' has no attribute 'I2C2_SCL_M0'. Did you mean: 'I2C2_SCL'?
>>> 
gpio readall
 +------+-----+----------+--------+---+ PI5 PLUS +---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 |      |     |     3.3V |        |   |  1 || 2  |   |        | 5V       |     |      |
 |   16 |   0 |    SDA.2 |   ALT9 | 1 |  3 || 4  |   |        | 5V       |     |      |
 |   15 |   1 |    SCL.2 |   ALT9 | 1 |  5 || 6  |   |        | GND      |     |      |
 |   62 |   2 |    PWM14 |     IN | 1 |  7 || 8  | 0 | IN     | GPIO1_A1 | 3   | 33   |
 |      |     |      GND |        |   |  9 || 10 | 0 | IN     | GPIO1_A0 | 4   | 32   |
 |   36 |   5 | GPIO1_A4 |     IN | 0 | 11 || 12 | 1 | IN     | GPIO3_A1 | 6   | 97   |
 |   39 |   7 | GPIO1_A7 |     IN | 1 | 13 || 14 |   |        | GND      |     |      |
 |   40 |   8 | GPIO1_B0 |     IN | 1 | 15 || 16 | 1 | IN     | GPIO3_B5 | 9   | 109  |
 |      |     |     3.3V |        |   | 17 || 18 | 0 | IN     | GPIO3_B6 | 10  | 110  |
 |   42 |  11 | SPI0_TXD |     IN | 0 | 19 || 20 |   |        | GND      |     |      |
 |   41 |  12 | SPI0_RXD |     IN | 0 | 21 || 22 | 1 | OUT    | GPIO1_A2 | 13  | 34   |
 |   43 |  14 | SPI0_CLK |     IN | 0 | 23 || 24 | 1 | IN     | SPI0_CS0 | 15  | 44   |
 |      |     |      GND |        |   | 25 || 26 | 1 | IN     | SPI0_CS1 | 16  | 45   |
 |   47 |  17 | GPIO1_B7 |     IN | 1 | 27 || 28 | 1 | IN     | GPIO1_B6 | 18  | 46   |
 |   63 |  19 | GPIO1_D7 |     IN | 1 | 29 || 30 |   |        | GND      |     |      |
 |   96 |  20 | GPIO3_A0 |     IN | 1 | 31 || 32 | 0 | IN     | GPIO1_A3 | 21  | 35   |
 |  114 |  22 | GPIO3_C2 |     IN | 0 | 33 || 34 |   |        | GND      |     |      |
 |   98 |  23 | GPIO3_A2 |     IN | 1 | 35 || 36 | 0 | IN     | GPIO3_A5 | 24  | 101  |
 |  113 |  25 | GPIO3_C1 |     IN | 0 | 37 || 38 | 0 | IN     | GPIO3_A4 | 26  | 100  |
 |      |     |      GND |        |   | 39 || 40 | 1 | IN     | GPIO3_A3 | 27  | 99   |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+ PI5 PLUS +---+--------+----------+-----+------+

sudo i2cdetect -y 2
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
00x48 is the address of the ads1115  A2D converter.

Am I right the blinka doesn't import the board and pin?
Thank you

drozdi70, 2024-03-10:
HiPlease follow https://github.com/drozdi70/openplotter_opi5plus/blob/main/p13.sh at the end of that file 2 files need to be copied. Tryb this...KindlyTemat: Re: [adafruit/Adafruit_Blinka] Orange PI 5 PLUS incorrect RK3388 definition (Issue #692)Data: 2024-03-08 10:38Nadawca: "3diamond" ***@***.***&gt;Adresat: "adafruit/Adafruit_Blinka" ***@***.***&gt;; DW: "DroZDi" ***@***.***&gt;; "Author" ***@***.***&gt;; 

add to blinka_test.py:
print("Board id is: ",board_id)
and provide the output.
and next try this (please make copy of files: pin.py, orangepi5.py, orangei5plus.py before you start):
if you see ORANGE_PI_5 you need to
sudo cp -f orangepi5.py /home/hmc/Code/.venv/lib/python3.10/site-packages/adafruit_blinka/board/orangepi/orangepi5.py
else ORANGE_PI_5_PLUS
sudo cp -f orangepi5plus.py /home/hmc/Code/.venv/lib/python3.10/site-packages/adafruit_blinka/board/orangepi/orangepi5plus.py sudo cp -p pin5plus.py /home/hmc/Code/.venv/lib/python3.10/site-packages/adafruit_blinka/microcontroller/rockchip/rk3588/pin.py
maybe it can help you :) files related you can find on my github repositories...

Hi
Thank you all for for your efforts. Tried to get my project to work for 2 months without disturbing you guys. I really need your expertise and help. I am new to adafruit and blinka. I tried to follow the method above but I can't seem to get it to work.
I am on /dev/i2c-2 (I2C2_M0) with:
3.3v Pin 1
SDA pin 3
SCL pin 5
GND pin 6
Here are the results I get:
Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.

import board
import digitalio
import busio
import sys
import adafruit_platformdetect.constants.boards as ap_board
from adafruit_blinka.agnostic import board_id, detector
print("Hello blinka!")
Hello blinka!
print("Chip id: ", detector.chip.id)
Chip id:  RK3588
print("Board id is: ",board_id)
Board id is:  ORANGE_PI_5_PLUS
i2c = busio.I2C(board.SCL, board.SDA)
Traceback (most recent call last):
File "", line 1, in 
File "/usr/local/lib/python3.10/dist-packages/busio.py", line 37, in init
self.init(scl, sda, frequency)
File "/usr/local/lib/python3.10/dist-packages/busio.py", line 162, in init
raise ValueError(
ValueError: No Hardware I2C on (scl,sda)=((0, 15), (0, 16))
Valid I2C ports: ((0, (4, 21), (4, 22)), (1, (0, 13), (0, 14)), (1, (1, 9), (1, 10)), (3, (3, 15), (3, 16)), (5, (1, 14), (1, 15)), (7, (4, 10), (4, 11)), (8, (3, 18), (3, 19)))
i2c = busio.I2C(board.I2C2_SCL_M0, board.I2C2_SDA_M0)
Traceback (most recent call last):
File "", line 1, in 
AttributeError: module 'board' has no attribute 'I2C2_SCL_M0'. Did you mean: 'I2C2_SCL'?

gpio readall
+------+-----+----------+--------+---+ PI5 PLUS +---+--------+----------+-----+------+
| GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
+------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
|      |     |     3.3V |        |   |  1 || 2  |   |        | 5V       |     |      |
|   16 |   0 |    SDA.2 |   ALT9 | 1 |  3 || 4  |   |        | 5V       |     |      |
|   15 |   1 |    SCL.2 |   ALT9 | 1 |  5 || 6  |   |        | GND      |     |      |
|   62 |   2 |    PWM14 |     IN | 1 |  7 || 8  | 0 | IN     | GPIO1_A1 | 3   | 33   |
|      |     |      GND |        |   |  9 || 10 | 0 | IN     | GPIO1_A0 | 4   | 32   |
|   36 |   5 | GPIO1_A4 |     IN | 0 | 11 || 12 | 1 | IN     | GPIO3_A1 | 6   | 97   |
|   39 |   7 | GPIO1_A7 |     IN | 1 | 13 || 14 |   |        | GND      |     |      |
|   40 |   8 | GPIO1_B0 |     IN | 1 | 15 || 16 | 1 | IN     | GPIO3_B5 | 9   | 109  |
|      |     |     3.3V |        |   | 17 || 18 | 0 | IN     | GPIO3_B6 | 10  | 110  |
|   42 |  11 | SPI0_TXD |     IN | 0 | 19 || 20 |   |        | GND      |     |      |
|   41 |  12 | SPI0_RXD |     IN | 0 | 21 || 22 | 1 | OUT    | GPIO1_A2 | 13  | 34   |
|   43 |  14 | SPI0_CLK |     IN | 0 | 23 || 24 | 1 | IN     | SPI0_CS0 | 15  | 44   |
|      |     |      GND |        |   | 25 || 26 | 1 | IN     | SPI0_CS1 | 16  | 45   |
|   47 |  17 | GPIO1_B7 |     IN | 1 | 27 || 28 | 1 | IN     | GPIO1_B6 | 18  | 46   |
|   63 |  19 | GPIO1_D7 |     IN | 1 | 29 || 30 |   |        | GND      |     |      |
|   96 |  20 | GPIO3_A0 |     IN | 1 | 31 || 32 | 0 | IN     | GPIO1_A3 | 21  | 35   |
|  114 |  22 | GPIO3_C2 |     IN | 0 | 33 || 34 |   |        | GND      |     |      |
|   98 |  23 | GPIO3_A2 |     IN | 1 | 35 || 36 | 0 | IN     | GPIO3_A5 | 24  | 101  |
|  113 |  25 | GPIO3_C1 |     IN | 0 | 37 || 38 | 0 | IN     | GPIO3_A4 | 26  | 100  |
|      |     |      GND |        |   | 39 || 40 | 1 | IN     | GPIO3_A3 | 27  | 99   |
+------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
| GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
+------+-----+----------+--------+---+ PI5 PLUS +---+--------+----------+-----+------+
sudo i2cdetect -y 2
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
00x48 is the address of the ads1115  A2D converter.
Am I right the blinka doesn't import the board and pin?
Thank you

3diamond, 2024-03-11:
Thank you. I followed most of the instructions and it works. Dziękuję.