adafruit/Adafruit_Blinka #696: Blinka's usb_hid.py with MicroPython throws ImportErrors

evgeni-student, 2023-08-25:
Hello! I have the latest MicroPython 1.20.0 installed on Raspberry Pi Pico H and I want to use CircuitPython the usb_hid.py module from Adafruit_Blinka.
I have the following directories in the lib folder and a sample MicroPython and CircuitPython mixed code runs as intended.
[image]
The issue I have appears when I try to import usb_hid with the intention to use adafruit_hid.keyboard, adafruit_hid.keycode, adafruit_hid.keyboard_layout_us and adafruit_hid.consumer_control.
[image]
I already seen these issues [#586](https://github.com/adafruit/Adafruit_Blinka/pull/586) and [#230](https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/230). I tried adding try/catch on the inner code but it just results in other errors like these:
`
File "/lib/usb_hid.py", line 19, in
ImportError: no module named 'pathlib'
`
and 
` 
ImportError: no module named 'atexit'
`
After installing typing from Thonny's manage package interface pictured below:
[image]
and running the code again, the following ImportError occurs:
[image]
further still when I change the import to just import typing and then wrapping the `from abc import abstractmethod, abstractproperty` part in try/catch I again bump into more problems like the one below:
[image]

ladyada, 2023-08-27:
the failures shouldn't be coming up but im also pretty sure the hid library wont run because micropython does not have core HID support for RP2040 - if you want to act like a keyboard you'll need to use circuitpy

evgeni-student, 2023-08-28:
> the failures shouldn't be coming up but im also pretty sure the hid library wont run because micropython does not have core HID support for RP2040 - if you want to act like a keyboard you'll need to use circuitpy

Thank you for your comment. Yes, when I use CircuitPython only it works, but my intention behind using MicroPython is to utilize hard interrupt handlers rather than polling a CircuitPython while loop.

makermelissa, 2023-08-29:
> Thank you for your comment. Yes, when I use CircuitPython only it works, but my intention behind using MicroPython is to utilize hard interrupt handlers rather than polling a CircuitPython while loop.

Looking through their documentation, it looks like the only board they added it to was the pyboard, which is also supported by Blinka.

evgeni-student, 2023-08-30:
I guess I would just have to wait until MicroPython gets native HID support for RP2040. Thank you for your help.