adafruit/Adafruit_Blinka #699: Support for NanoPi Neo3 pls

liam-harrington, 2023-09-07:
https://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO3

PLS add example for adding new boards <3

ruskifilozof, 2024-03-17:
Please help with adding support for Nano Pi NEO 3. Adafruit-PlatformDetect version 3.62.0 detecting chip RK3328 correctly but "self._rock_pi_id()" returns "None", in result whole platform detection fails. I've tried to add support using link above but stuck. Any chance to get this board working? 
Thank you in advance.

makermelissa, 2024-03-18:
@ruskifilozof yeah, the board still hasn't been added yet. if you are seeing the chip being detected correct, then that's some of the work already done for you. If you'd like to get it working, you can follow the guide that @jepler linked to above in this issue.