adafruit/Adafruit_Blinka #706: import board command not working

kylecollins9975, 2023-09-11:
### Board Name

ESP 32-S Feather

### Steps

1. connect board to computer
2. run import board code

### Description

Adafruit-PlatformDetect version 3.52.0 was unable to identify the board and/or
        microcontroller running the Darwin platform

makermelissa, 2023-09-11:
Which board is this for? There are a number of feathers with the ESP32. I presume you are requesting adding support for either the ESP32 S2 or S3.

kylecollins9975, 2023-09-11:
Hi, apologies for the confusion. I am reffering to the S2, thanks for the help.

makermelissa, 2026-04-03:
In your initial post, it mentions running it on Darwin (MacOS). Just to be clear, the only way this can run on the ESP32-S2 Feather is on top of MicroPython. In most cases, it would be better to just use CircuitPython.