adafruit/Adafruit_Blinka #717: Blinka library usb_hid fails host enumeration with simple keyboard example

eightycc, 2023-09-16:
### Board Name

Raspberry Pi Zero W

### Steps

1. Install latest 32-bit Raspbian on Zero and update: `Linux zero 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
2. Setup the Zero with a Pi Cobbler and wire up 2 tactile switches and an LED according to standard practice
3. Install Blinka 8.22.3 according to these instructions: https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi
4. Add the line `dtoverlay=dwc2` to Zero's `/boot/config.txt`
5. Reboot Zero
6. Modify the simple HID keyboard demo from this page https://learn.adafruit.com/circuitpython-essentials/circuitpython-hid-keyboard-and-mouse to use tactile switches and LED from step 2 above
7. Copy library `adafruit-circuitpython-hid-py-5.4.0/lib/adafruit_hid` to `~/.local/lib/python3.9/site-packages/`
8. `sudo modprobe libcomposite`
9. Connect Zero to a port on a Linux box. I use an adapter with V+ cut so that I can keep the Zero powered and avoid lengthy reboots, but problem will also reproduce with the Linux box powering the Zero
10. Run the HID keyboard demo `python3 key_hid_example.py`

On the Zero:
```
rabeles@zero:~ $ python3 key_hid_example.py 
Traceback (most recent call last):
  File "/home/rabeles/key_hid_example.py", line 27, in <module>
    keyboard = Keyboard(usb_hid.devices)
  File "/home/rabeles/.local/lib/python3.9/site-packages/adafruit_hid/keyboard.py", line 49, in __init__
    self._keyboard_device = find_device(devices, usage_page=0x1, usage=0x06)
  File "/home/rabeles/.local/lib/python3.9/site-packages/adafruit_hid/__init__.py", line 47, in find_device
    raise ValueError("Could not find matching HID device.")
ValueError: Could not find matching HID device.
```

dmesg output on the Linux box:
```
[177927.142463] usb 1-2.1: device descriptor read/64, error -110
[177927.250776] usb 1-2-port1: attempt power cycle
[177927.854441] usb 1-2.1: new full-speed USB device number 43 using xhci_hcd
[177930.016489] usb 1-2.1: Device not responding to setup address.
[177930.228693] usb 1-2.1: Device not responding to setup address.
[177930.434453] usb 1-2.1: device not accepting address 43, error -71
[177988.671121] usb 1-2.1: new full-speed USB device number 45 using xhci_hcd
```

### Description

Since I will need this working to test resolution to #711, I'll be looking into this issue first.

eightycc, 2023-09-16:
@dhalbert Any thoughts on this?

eightycc, 2023-09-19:
Because the `usb_hid` implementation depends on `USB gadget` in the Linux kernel, it needs to run as `superuser`. As a consequence of this, Blinka needs to be installed using `sudo pip3 install --upgrade adafruit-blinka` and `adafruit_hid` needs to be installed into `/usr/lib/python3.9/site-packages`.

eightycc, 2023-09-19:
@makermelissa `usb_hib` could use some help. It looks like the documentation is missing, and an example would be nice. Let me see what I can do. Is there an issue for Neopixel?

makermelissa, 2023-09-19:
#628 is already an issue for adding documentation for usb_hid and the issue for NeoPixels is #381

eightycc, 2023-09-19:
With a bit more coaxed, my simple keyboard hid is working. The root problem is lack of documentation. Closing