adafruit/Adafruit_Blinka #726: FT232H Error: Notimplementederror

APAKGRUP, 2023-10-20:
**Board Name:** 
FT232H

**Steps:**

1. connect board to computer
2. install all requirements and check them
3. run this simple code:
import board
import digitalio
import time

led = digitalio.DigitalInOut(board.C0)
Buzzer = digitalio.DigitalInOut(board.C1)

led.direction = digitalio.Direction.OUTPUT
Buzzer.direction = digitalio.Direction.OUTPUT

while True:
    
    
    led.value = True
    Buzzer.value = True
    time.sleep(0.5)
    led.value = False
    Buzzer.value = False
    
**Description:**
NotImplementedError: 
        Adafruit-PlatformDetect version 3.53.0 was unable to identify the board and/or
        microcontroller running the Windows platform. Please be sure you
        have the latest packages by running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

        If you are running the latest package, your board may not yet be supported. Please
        open a New Issue on GitHub at https://github.com/adafruit/Adafruit_Blinka/issues and
        select New Board Request.
**Additional information**
I tried to install all the requirements and extra libraries. Still goting the same error

caternuson, 2023-10-20:
> 2. install all requirements and check them

Did they all pass?
https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/troubleshooting

APAKGRUP, 2023-10-23:
> > 2. install all requirements and check them
> 
> Did they all pass? https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/troubleshooting

Hello,
When I checked the environment variable within Python I got a KeyError, which means I did not set the environment variable right.
Do you know how I can fix this?
Thanks

caternuson, 2023-10-23:
Set the environment variable:
https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/windows#set-environment-variable-3041721

APAKGRUP, 2023-10-24:
update: I changed env variable from my Python code by using this code:

import os

os.environ["BLINKA_FT232H"] = "1"

import board
Still getting the same error

caternuson, 2023-10-24:
What if you try it as shown in guide?
https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/windows#set-environment-variable-3041721

APAKGRUP, 2023-10-26:
> What if you try it as shown in guide? https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/windows#set-environment-variable-3041721

Hello dear, 
I mentioned that I completed all the steps correctly on cmd. But when I ran my code on Spyder the same error message appeared. 
Thanks

caternuson, 2023-10-26:
Something is preventing Spyder (or some other aspect of your setup) from seeing the environment variable settings. It's unclear how it sees it with the check done here:
https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/troubleshooting#check-environment-variable-within-python-3041702
but not when importing board. The underlying system relies on the environment variable to know that a FT232H is in use:
https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/bb2d27494d8556d8f7fa4b8b0d48c1ce37ba85fd/adafruit_platformdetect/chip.py#L65

This is something unique to your setup.

closed 2023-10-26 by makermelissa