adafruit/Adafruit_Blinka #730: Jetson Orin Nano no SCLK_1

SmartAnder, 2023-10-24:
### Board Name

Jetson Orin Nano

### Steps

Hi, I am trying to get a TFT screen with an ILI9341 driver to work using the second SPI port of the Orin Nano since the first one is blocked with another device. When I use the provided drivers with SPI0(pin19...) from https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display it works great.

When I change the pins to their second SPI and change the code accordingly ->[rgb.zip](https://github.com/adafruit/Adafruit_Blinka/files/13086385/rgb.zip)
 I get the following error
"""
Event not found
  warnings.warn("Event not found", RuntimeWarning)
Traceback (most recent call last):
  File "rgb.py", line 17, in <module>
    spi = busio.SPI(board.SCK_1, MOSI=board.MOSI_1, MISO=board.MISO_1)
AttributeError: module 'board' has no attribute 'SCK_1'
"""

I changed those lines as specified in https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano/spi-sensors-devices 
[image]

I have gone through the busio libraries to see if it could be a typo and there is no reference to SCK_1 only SCK. I have also looked into the circuit python documentation https://docs.circuitpython.org/en/latest/shared-bindings/busio/index.html#busio.SPI and there is no reference to it

Could you guys help me solve this? Thanks

### Description

-might be related to #110  and #148 but those are from 2019

makermelissa, 2023-10-24:
@anhmiuhv any ideas? I'm not familiar with the pinout of the Jetson Orin Nano.

anhmiuhv, 2023-10-25:
I believe this article
https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano/spi-sensors-devices

Refer to Jetson Nano not Jetson Orin Nano

makermelissa, 2023-10-25:
Thank you @anhmiuhv. That makes more sense. Closing since the Jetson Nano does have SCK_1.

closed 2023-10-25 by makermelissa