adafruit/Adafruit_Blinka #737: ft232h with MLX90640

APAKGRUP, 2023-11-01:
**Board Name**
ft232h

**Steps**

1. Connect MLX90640 with the board as following: VCC=5V,GND=GND,SDA=D1,SCL=D0
2. Connect the board to the computer
3. Run the usage example code on this page: https://github.com/adafruit/Adafruit_CircuitPython_MLX90640

**Description**
valueerror: no i2c device at address: 0x33

**Additional information**
I am connecting ft232h to a PC that has a Windows 10 operating system.

caternuson, 2023-11-01:
Are you using an Adafruit MLX90640 breakout?
https://www.adafruit.com/product/4407

caternuson, 2023-11-02:
Not using Adafruit MLX90640 breakout:
https://forums.adafruit.com/viewtopic.php?p=991425

Please contact the maker of the MLX90640 board for support with their product.

closed 2023-11-02 by caternuson