adafruit/Adafruit_Blinka #747: rpi_ws281x not compatible with pi 5

mundo1979, 2023-12-01:
### Board Name

Raspberry Pi 5

### Steps

Output of example from here- https://learn.adafruit.com/neopixels-on-raspberry-pi/python-usage:

  File "/root/test.py", line 64, in <module>
    pixels.show()
  File "/root/env/lib/python3.11/site-packages/adafruit_pixelbuf.py", line 204, in show
    return self._transmit(self._post_brightness_buffer)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/env/lib/python3.11/site-packages/neopixel.py", line 180, in _transmit
    neopixel_write(self.pin, buffer)
  File "/root/env/lib/python3.11/site-packages/neopixel_write.py", line 42, in neopixel_write
    return _neopixel.neopixel_write(gpio, buf)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/neopixel.py", line 83, in neopixel_write
    raise RuntimeError(
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
swig/python detected a memory leak of type 'ws2811_t *', no destructor found.

### Description

The error appears to be from rpi_ws281x

The issue is currently being worked on:
https://github.com/jgarff/rpi_ws281x/issues/528

stephinson, 2023-12-29:
@makermelissa I also faced a similar problem and tried SPI, as you suggested. See [here](https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel_SPI/issues/37#issuecomment-1871698900)

makermelissa, 2023-12-29:
Another option to try if you don't want to wait for the driver to be fixed, is to have use a NeoDriver (https://www.adafruit.com/product/5766) to separately control the lights.

stephinson, 2023-12-30:
> Another option to try if you don't want to wait for the driver to be fixed, is to have use a NeoDriver (https://www.adafruit.com/product/5766) to separately control the lights.

I bought an Arduino Uno instead for now

Nayan96, 2025-05-14:
Hi @makermelissa, is there any update on this issue? From what I understand, the main blocker might be changes in the DMA or PWM hardware interface on the Pi 5 — is that right? Or is something else in the way?

makermelissa, 2025-05-14:
Oh yeah, Neopixels should work on the Pi 5 now. It doesn't use rpi_ws281x, but some PIO code. It was added in https://github.com/adafruit/Adafruit_Blinka/pull/917.

closed 2025-05-14 by makermelissa

Nayan96, 2025-05-14:
Oh awesome, thanks for the update!