adafruit/Adafruit_Blinka #752: Raspberry Pi5 RuntimeError: Timed out waiting for PulseIn message. Make sure libgpiod is installed.

eMUQI, 2023-12-12:
### Board Name

Raspberry Pi5 

### Steps

Hi there! Recently, I've been following the guide at https://learn.adafruit.com/dht-humidity-sensing-on-raspberry-pi-with-gdocs-logging/python-setup to work with my DHT11 sensor using Blinka on my Raspberry Pi 5.
1. To start off, I followed the instructions at https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi and executed the commands below step by step:
``` bash
sudo apt-get update
sudo apt-get upgrade
sudo apt install python3.11-venv
python -m venv env --system-site-packages
source env/bin/activate
``` 
``` bash
cd ~
pip3 install --upgrade adafruit-python-shell
wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/raspi-blinka.py
sudo -E env PATH=$PATH python3 raspi-blinka.py
```
2. Then I restarted my Raspberry Pi according to the prompt and ran the code below
``` python
import board
import digitalio
import busio

print("Hello blinka!")

# Try to great a Digital input
pin = digitalio.DigitalInOut(board.D4)
print("Digital IO ok!")

# Try to create an I2C device
i2c = busio.I2C(board.SCL, board.SDA)
print("I2C ok!")

# Try to create an SPI device
spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
print("SPI ok!")

print("done!")
```
And get:
```
Hello blinka!
Digital IO ok!
I2C ok!
SPI ok!
done!
```
Everything above went as expected, very smoothly.
3. However, when I ran the sample code in this tutorial https://learn.adafruit.com/dht-humidity-sensing-on-raspberry-pi-with-gdocs-logging/python-setup, I encountered an error.
``` bash
pip3 install adafruit-circuitpython-dht
sudo apt-get install libgpiod2
```

``` python
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import time
import board
import adafruit_dht

# Initial the dht device, with data pin connected to:
dhtDevice = adafruit_dht.DHT11(board.D17)

# you can pass DHT22 use_pulseio=False if you wouldn't like to use pulseio.
# This may be necessary on a Linux single board computer like the Raspberry Pi,
# but it will not work in CircuitPython.
# dhtDevice = adafruit_dht.DHT22(board.D18, use_pulseio=False)

while True:
    try:
        # Print the values to the serial port
        temperature_c = dhtDevice.temperature
        temperature_f = temperature_c * (9 / 5) + 32
        humidity = dhtDevice.humidity
        print(
            "Temp: {:.1f} F / {:.1f} C    Humidity: {}% ".format(
                temperature_f, temperature_c, humidity
            )
        )

    except RuntimeError as error:
        # Errors happen fairly often, DHT's are hard to read, just keep going
        print(error.args[0])
        time.sleep(2.0)
        continue
    except Exception as error:
        dhtDevice.exit()
        raise error

    time.sleep(2.0)

```

```
invalid GPIO offset: (4, 17)Traceback (most recent call last):
  File "/home/pi/Code/dht11.py", line 9, in <module>
    dhtDevice = adafruit_dht.DHT11(board.D17)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/env/lib/python3.11/site-packages/adafruit_dht.py", line 294, in __init__
    super().__init__(True, pin, 18000, use_pulseio)
  File "/home/pi/env/lib/python3.11/site-packages/adafruit_dht.py", line 86, in __init__
    self.pulse_in = PulseIn(self._pin, maxlen=self._max_pulses, idle_state=True)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 85, in __init__
    message = self._wait_receive_msg(timeout=0.25)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 103, in _wait_receive_msg
    raise RuntimeError(
RuntimeError: Timed out waiting for PulseIn message. Make sure libgpiod is installed.
```

I looked into some potentially related issues, such as those found at https://github.com/adafruit/Adafruit_Blinka/issues/740 and https://github.com/adafruit/Adafruit_Blinka/issues/259. However, I still couldn't resolve my problem.  Also, I'm pretty sure that I installed `libgpiod` by running `sudo apt-get install libgpiod2`.

I would appreciate any insights or suggestions on troubleshooting this issue. Has anyone else encountered this problem with the Raspberry Pi 5, and if so, how was it resolved?

Thank you for your time and help!

### Additional information

Raspberry Pi 5 Model B Rev 1.0
OS: Debian GNU/Linux 12 (bookworm) aarch64
Kernel: 6.1.0-rpi7-rpi-2712

makermelissa, 2023-12-27:
I think the pulsein binary likely needs to be be recompiled for the Pi 5.

makermelissa, 2023-12-27:
It didn't need recompiling, it just needed to use the correct gpiochip and offset.

closed 2023-12-29 by makermelissa

eMUQI, 2024-01-31:
It works great now, thanks!

gkellershs, 2024-02-14:
Hello,
I have a similar situation, Pi5 and the DHT22 will not work. What have to be done to get it running?

I get the following error
```
(env) pi@instructorPI5:~ $ python3 dht.py
<module 'board' from '/home/pi/.local/lib/python3.11/site-packages/board.py'>
Traceback (most recent call last):
  File "/home/pi/dht.py", line 12, in <module>
    dhtDevice = Adafruit_DHT.DHT22(board.D6, use_pulseio=False)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'int' object is not callable
```

This is the code snipplet:

```
import time
import board
import Adafruit_DHT
print (board)
# Initial the dht device, with data pin connected to:
dhtDevice = Adafruit_DHT.DHT22(board.D6, use_pulseio=False)
```

makermelissa, 2024-02-14:
@gkellershs I think `adafruit_dht` may need to be lowercase. See this example: https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/main/examples/dht_simpletest.py

gkellershs, 2024-02-14:
I installed it on a fresh image and now it´s working, thank you very much.

PR #764 (2023-12-29): Fix pulsein for the Pi 5 and make lockable silent
Fixes #748. Fixes #752. Tested on Pi 5.