adafruit/Adafruit_Blinka #759: RP2040 fails on valid UART pin combinations

shawnferry, 2023-12-18:
### Board Name

rp2040 on custom PCB

### Steps

Define UART on valid pins

[image]

```
busio.UART(board.GP16, board.GP17)
```

```
E   ValueError: No Hardware UART on (tx,rx)=(16, 17)
E   Valid UART ports: ((0, board.GP0, board.GP1), (0, board.GP0, board.GP13), (0, board.GP12, board.GP1), (0, board.GP12, board.GP13), (1, board.GP4, board.GP5), (1, board.GP4, board.GP9), (1, board.GP8, board.GP5), (1, board.GP8, board.GP9))
```

### Additional information

Pull request incoming

makermelissa, 2023-12-19:
It sounds like you are running it as one of the U2IF boards. What board definition are you using?

makermelissa, 2023-12-19:
Never mind, looks like you already figured out the issue and submitted #760.

PR #760 (2023-12-19): Add missing UART definitions from RP2040 datasheet (a6fe703-clean)
Continue implementation to match i2c and SPI 

See https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf  1.4.3. GPIO Functions

Fixes #759
closed 2023-12-19 by makermelissa