adafruit/Adafruit_Blinka #778: Is the Rock 4SE not supported?

A-Little-on-the-Nose, 2024-01-18:
Hello,
I am trying to use the adafruit-blanka and adafruit-trellis library for my Rock 4 SE, but there seems to be an error with libgpiod.

Traceback (most recent call last):
  File "main.py", line 1, in <module>
    import start_window
  File "/home/.../Launchpad/start_window.py", line 3, in <module>
    from settings_window import settings_window
  File "/home/.../Launchpad/settings_window.py", line 2, in <module>
    import one_trellis_window
  File "/home/.../Launchpad/one_trellis_window.py", line 6, in <module>
    import board
  File "/home/.../.local/lib/python3.8/site-packages/board.py", line 269, in <module>
    from adafruit_blinka.board.radxa.rockpi4 import *
  File "/home/.../.local/lib/python3.8/site-packages/adafruit_blinka/board/radxa/rockpi4.py", line 7, in <module>
    from adafruit_blinka.microcontroller.rockchip.rk3399 import pin
  File "/home/.../.local/lib/python3.8/site-packages/adafruit_blinka/microcontroller/rockchip/rk3399/pin.py", line 7, in <module>
    from adafruit_blinka.microcontroller.generic_linux.libgpiod_pin import Pin
  File "/home/.../.local/lib/python3.8/site-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 8, in <module>
    raise ImportError(
ImportError: libgpiod Python bindings not found, please install and try again! See https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/libgpiod.py

makermelissa, 2024-01-18:
It doesn't look like the Rock 4 SE has been added (See https://circuitpython.org/blinka?q=Rock). However, the error indicates that libgpiod python bindings haven't been installed. You could try
`sudo apt install python3-libgpiod` or `pip install gpiod` and see if it works after that.

A-Little-on-the-Nose, 2024-01-19:
The thing that helped was to undo the changes of the pin.py to an older version. But could it be that I can us this library only in combination with trellis plus rock pi and not with trellis->arduino->rock 4SE?

makermelissa, 2024-02-26:
If you would like to add support for the Rock 4 SE by modifying the pin file without breaking whichever board you are updating, that would be helpful. I'm not sure exactly what you are trying to do, but with this library, you should be able to interact with the trellis directly. It's basically a combination of using the keypad module and neopixels. I mean, it's theoretically possible to do it with using the arduino as a translation layer, but it sounds pretty advanced.

makermelissa, 2024-02-28:
Fixed by #791.

closed 2024-02-28 by makermelissa