adafruit/Adafruit_Blinka #782: SSD1306 is not working on a Raspberry Pi 5

Martius108, 2024-01-23:
### Board Name

Raspberry Pi 5

### Steps

Running an example script to display something on a SSD1306 display. See details below.

### Description

Might be related to #776 

I followed this tutorial and created a virtual environment for the installation of:
pip3 install adafruit-circuitpython-ssd1306

Running the example script from here:
https://learn.adafruit.com/monochrome-oled-breakouts/python-usage-2

Leads to this error message:

`(project) maddin@maddin-desktop:~/project$ python3 ssd1306_test.py
ImportError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/maddin/project/ssd1306_test.py", line 1, in <module>
    import board
  File "/home/maddin/project/project/lib/python3.11/site-packages/board.py", line 50, in <module>
    from adafruit_blinka.board.raspberrypi.raspi_5b import *
  File "/home/maddin/project/project/lib/python3.11/site-packages/adafruit_blinka/board/raspberrypi/raspi_5b.py", line 6, in <module>
    from adafruit_blinka.microcontroller.bcm2712 import pin
  File "/home/maddin/project/project/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm2712/pin.py", line 5, in <module>
    from adafruit_blinka.microcontroller.generic_linux.libgpiod_pin import Pin
  File "/home/maddin/project/project/lib/python3.11/site-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 8, in <module>
    raise ImportError(
ImportError: libgpiod Python bindings not found, please install and try again! See https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/libgpiod.py`

makermelissa, 2024-01-23:
Try installing with `pip3 install gpiod`.

As an aside, the message probably needs to be updated as the libgpiod script in most cases is unnecessary (maybe it should direct users to a learn guide instead). Also gpiod may need to be specified as a dependency so that it's installed automatically.

Martius108, 2024-01-24:
Thank you. I installed gpiod.

After starting the example script I get:

`Traceback (most recent call last):
  File "/home/maddin/project/ssd1306_test.py", line 2, in <module>
    import displayio
ModuleNotFoundError: No module named 'displayio'
(project) maddin@maddin-desktop:~/project$ `

Can I skip the import of displayio or do I have to import another module/library?

Do you have a less sophisticated test script for the SSD1306 display?

Martius108, 2024-01-26:
I asked a KI to write me a suitable test script with gpiod a this is what I got:

`import gpiod

def main():
    # Initialisiere die I2C-Schnittstelle
    bus = gpiod.I2CBus(port=1)

    # Liste aller I2C-Geräte ausgeben
    for address in range(0x00, 0x7F):
        device = bus.get_device(address)
        if device is not None:
            print(f"Gerät gefunden: {address} {device.name}")

if __name__ == "__main__":
    main()`

but it doesn’t work :)

eMUQI, 2024-02-01:
On my Raspberry Pi 5, the SSD1306 works fine.

```
OS: Debian GNU/Linux 12 (bookworm) aarch64
Host: Raspberry Pi 5 Model B Rev 1.0
Kernel: 6.1.0-rpi7-rpi-2712
```
OLED from [sunfounder](https://www.sunfounder.com/products/0-96-iic-i2c-serial-128x64-ssd1315-oled-display-module) 

Maybe you can refer to my process, it may be helpful for you.

I followed the tutorial at https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/overview to install blinka.
Then installed adafruit-circuitpython-ssd1306.

Run the following code:
``` python
# https://learn.adafruit.com/monochrome-oled-breakouts/python-usage-2

# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

"""
This demo will fill the screen with white, draw a black box on top
and then print Hello World! in the center of the display

This example is for use on (Linux) computers that are using CPython with
Adafruit Blinka to support CircuitPython libraries. CircuitPython does
not support PIL/pillow (python imaging library)!
"""

import board
import digitalio
from PIL import Image, ImageDraw, ImageFont
import adafruit_ssd1306

# Change these
# to the right size for your display!
WIDTH = 128
HEIGHT = 64  # Change to 64 if needed
BORDER = 5

# Use for I2C.
i2c = board.I2C()  # uses board.SCL and board.SDA
# i2c = board.STEMMA_I2C()  # For using the built-in STEMMA QT connector on a microcontroller
oled = adafruit_ssd1306.SSD1306_I2C(WIDTH, HEIGHT, i2c, addr=0x3C)

# Use for SPI
# spi = board.SPI()
# oled_cs = digitalio.DigitalInOut(board.D5)
# oled_dc = digitalio.DigitalInOut(board.D6)
# oled = adafruit_ssd1306.SSD1306_SPI(WIDTH, HEIGHT, spi, oled_dc, oled_reset, oled_cs)

# Clear display.
oled.fill(0)
oled.show()

# Create blank image for drawing.
# Make sure to create image with mode '1' for 1-bit color.
image = Image.new("1", (oled.width, oled.height))

# Get drawing object to draw on image.
draw = ImageDraw.Draw(image)

# Draw a white background
draw.rectangle((0, 0, oled.width, oled.height), outline=255, fill=255)

# Draw a smaller inner rectangle
draw.rectangle(
    (BORDER, BORDER, oled.width - BORDER - 1, oled.height - BORDER - 1),
    outline=0,
    fill=0,
)

# Load default font.
font = ImageFont.load_default()

# Draw Some Text
text = "Hello World!"
(font_width, font_height) = font.getsize(text)
draw.text(
    (oled.width // 2 - font_width // 2, oled.height // 2 - font_height // 2),
    text,
    font=font,
    fill=255,
)

# Display image
oled.image(image)
oled.show()
```

The example runs successfully with only one warning:
```
/home/pi/Code/oled_test.py:66: DeprecationWarning: getsize is deprecated and will be removed in Pillow 10 (2023-07-01). Use getbbox or getlength instead.
  (font_width, font_height) = font.getsize(text)
```

makermelissa, 2024-02-01:
> The example runs successfully with only one warning:
> 
> ```
> /home/pi/Code/oled_test.py:66: DeprecationWarning: getsize is deprecated and will be removed in Pillow 10 (2023-07-01). Use getbbox or getlength instead.
>   (font_width, font_height) = font.getsize(text)
> ```

Looks like the example for that library may need updating.

Another option to try this out is with this library https://github.com/adafruit/Adafruit_Blinka_Displayio and following the directions on https://learn.adafruit.com/monochrome-oled-breakouts/circuitpython-setup. I added OLED support support in September last year and tested it with the SSD1306.

Martius108, 2024-02-02:
Hi,

thanks for your help! The example script from eMUQI finally worked on my Pi 5.

makermelissa, 2024-03-21:
It looks like the code has been updated. I went through and updated/tested the guide and all is working.

closed 2024-03-21 by makermelissa