adafruit/Adafruit_Blinka #787: Adafruit-PlatformDetect version 3.60.0 was unable to identify the board

DorCa77, 2024-02-12:
Dear all,

Adafruit-PlatformDetect version 3.60.0 was unable to identify the board and/or
        microcontroller running the Darwin platform. Please be sure you
        have the latest packages by running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

Ok, I ran 'pip3 install --upgrade adafruit-blinka adafruit-platformdetect', but I still get the same message in RStudio.

        If you are running the latest package, your board may not yet be supported. Please
        open a New Issue on GitHub at https://github.com/adafruit/Adafruit_Blinka/issues and
        select New Board Request.

Best regards

mrdrdemmy, 2024-02-17:
Same deal here (yes, latest package is reported to have been installed).  At the Python prompt:

`>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\xyz\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\board.py", line 387, in <module>
    raise NotImplementedError(
NotImplementedError:
        Adafruit-PlatformDetect version 3.60.0 was unable to identify the board and/or
        microcontroller running the Windows platform. Please be sure you
        have the latest packages by running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'`

I am not sure what is being looked for here.  The desktop machine is a year-old Dell Vostro, running an up-to-date Windows 11.  A mcp2221a breakout is connected, and looks fine in device manager.  What's up?

raku9cam, 2024-02-19:
I had the same problem, but changing the Python version from 3.10 to 3.9 fixed it.

mrdrdemmy, 2024-02-22:
I have version 3.11.6 installed, and bristle at being compelled to roll back just to address an otherwise obscure issue.  I will give it a shot, though, and see whether it was worth the effort.  Thanks for the tip!

mrdrdemmy, 2024-02-22:
Well, that was easy.  On a whim, I moved up to 3.12.2, instead of down to 3.9, and the blinka install worked after that.  Thanks again for the versioning tip!

mrdrdemmy, 2024-02-22:
Whoops.  Upgrading to 3,12,2 **DID NOT** fix the problem;  _import board_ continues to yield the same error as above.  Python.org no longer offers an installer for version 3.9, so I will be putting this little project aside pending more ideas.  Too bad, since the mcp2221 breakout would have been pretty handy.

Apologies, if I led anyone astray with my previous comment!

dhalbert, 2024-02-22:
@makermelissa This appears to be failing with Python 3.12 on Windows and macOS, even with the MCP2221 env var set?

dhalbert, 2024-02-22:
@mrdrdemmy and @raku9cam Please edit your posts above to make clear what platform you are running on, if you've set any environment variables for an MCP2221, etc., and how you're setting them. Someone was trying to set the variable in a VScode terminal window, for instance, and that did not seem to work.

mrdrdemmy, 2024-02-23:
I did set BLINKA_MCP2221=1 in the VS Code terminal.  It's interesting to learn about the potential VS Code issue, though I haven't noticed trouble setting environment variables there before (e.g., when wrestling with the pico pi sdk install).  I'll try setting it outside of VS Code when I get a chance.  Thank you.

dhalbert, 2024-02-23:
@caternuson of interest to you

closed 2024-02-23 by dhalbert

dhalbert, 2024-02-23:
@mrdrdemmy Note that the Terminal window in VSCode is PowerShell, by default, so you need to use `$env:BLINKA_MCP2221=1`, _not_ `set BLINKA_MCP2221=1`.

Ney1ro, 2024-04-09:
**### _у меня пайчарм
 и выходит ошибка_**
 If you are running the latest package, your board may not yet be supported. Please
        open a New Issue on GitHub at https://github.com/adafruit/Adafruit_Blinka/issues and
        select New Board Request.