adafruit/Adafruit_Blinka #793: KB 2040 board

jjfcode, 2024-03-02:
Hello 
I'm having this issue

import board
NotImplementedError: 
        Adafruit-PlatformDetect version 3.61.0 was unable to identify the board and/or
        microcontroller running the Windows platform. Please be sure you
        have the latest packages by running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

Thank you for all your help
John

makermelissa, 2024-03-04:
Make sure your board is supported by Blinka. Only a. few RP2040 boards are: https://circuitpython.org/blinka?q=RP2040

jjfcode, 2024-03-04:
> Make sure your board is supported by Blinka. Only a. few RP2040 boards are: https://circuitpython.org/blinka?q=RP2040

ThankYou I will check on that

freemansoft, 2025-01-25:
Note that the boards in the link https://circuitpython.org/blinka?q=RP2040 don't seem to match the boards supported by U2IF https://github.com/adafruit/u2if/tree/0.1.1 that is probably required for blinka.