adafruit/Adafruit_Blinka #795: Support for linux (kernel > 4.8)

sriram-rs, 2024-03-18:
### Board Name

RockPiS

### Steps

1. SSH into the board
2. install adafruit_blinka library
3. open python and try to configure any gpio 

Adafruit blinka OSError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.11/dist-packages/digitalio.py", line 187, in __init__
    self.direction = Direction.INPUT
    ^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/digitalio.py", line 217, in direction
    self._pin.init(mode=Pin.IN)
  File "/usr/local/lib/python3.11/dist-packages/adafruit_blinka/microcontroller/generic_linux/sysfs_pin.py", line 94, in init
    self._open(self.IN)
  File "/usr/local/lib/python3.11/dist-packages/adafruit_blinka/microcontroller/generic_linux/sysfs_pin.py", line 142, in _open
    raise GPIOError(e.errno, "Exporting GPIO: " + e.strerror) from IOError
adafruit_blinka.microcontroller.generic_linux.sysfs_pin.GPIOError: [Errno 2] Exporting GPIO: No such file or directory

### Description

Error while using digitialio to configure GPIO with linux 6.1

Since the /sys/class/gpio is deprecated since kernel 4.8, Adafruit_Blinka cannot be used to configure and control gpios on the upstream kernel.

Now there is access only to /dev/gpiochip{x} instead of /sys/class/gpio/{x}.

Any help would be greatly appreciated.

makermelissa, 2024-03-18:
Sounds like it needs to be updated to use libgpiod. If I recall correctly, I'm the one who added this board and at the time libgpiod wasn't available, but if they are updating the kernel, perhaps it is now.

sriram-rs, 2024-03-19:
Thanks very much @makermelissa for the revert. Looking forward to the inclusion. So that I can use Adafruit_Blinka alone for all peripheral access with busio and IO access with digitalio.

makermelissa, 2024-03-20:
Got my RockPi S out and have now re-familiarized myself with it. Which image are you running? The official Rock Pi S Buster image is still on kernel 4.4.

makermelissa, 2024-03-20:
Ok, the official buster image is kinda terrible. It's kernel 4.4 and although libgpiod is available, there are no /dev/gpiochipX devices, so it's pointless. I'm going to update for Armbian.

sriram-rs, 2024-03-21:
Thanks @makermelissa again for seamlessly pointing out to the issue 👍🏽 . 

I am running 6.1.68 rc image provided by radxa.

closed 2024-03-21 by makermelissa

sriram-rs, 2024-03-22:
Thanks a lot, @makermelissa  and @tekktrik for the support. 

I will give the library a try and let know in case of queries. !!

PR #799 (2024-03-21): Update Rockchip rk3308 to use libgpiod
Fixes #795. Updates the Rockchip from using SysFs to libgpiod. Tested on Armbian.

To get this to work, the following needs to be done:
Setup a venv and activate:
```
sudo apt install python3-venv
python3 -m venv env --system-site-packages
source env/bin/activate
```

Then to give the user permissions without needing root access:
```
sudo groupadd gpiod
sudo usermod -G gpiod $USER
sudo nano /etc/udev/rules.d/60-gpiod.rules
```
Add the following to the file:
```
# udev rules for gpio port access through libgpiod
SUBSYSTEM=="gpio", KERNEL=="gpiochip0", GROU [... cut]