adafruit/Adafruit_Blinka #796: These are very inexpensive and I think would be good boards to support -- LuckyFox Pico Mini

earlthesquirrel, 2024-03-19:
https://www.spotpear.com/index/product/detail/id/1413.html Is a link to the board.
They are nice small boards that have a nice CPU and a fair number of GPIO and camera interface.

Would love to be able to use them with Circuit Python

makermelissa, 2024-03-19:
Oh nice! Yes, these would be great to add.

nopnop2002, 2024-06-07:
@earlthesquirrel 

Please wait a little more.

https://github.com/adafruit/Adafruit_Blinka/pull/841

nopnop2002, 2024-06-08:
@earlthesquirrel 

I found a problem and will fix it.

https://github.com/adafruit/Adafruit_Blinka/pull/842

[image]
[image]
[image]
[image]

makermelissa, 2024-07-18:
These have now been added.

closed 2024-07-18 by makermelissa

earlthesquirrel, 2024-08-21:
I see this marked as complete.  @makermelissa @nopnop2002 can you tell me if there is a CircuitPython release that includes this board?  I see some links and that BlinkA is available but it's not clear how to get CircuitPython running on this board.

nopnop2002, 2024-08-21:
@earlthesquirrel 

LuckFox comes with CPython installed.

dhalbert, 2024-08-21:
Since these run Linux, you would use "regular Python" and the Adafruit Blinka library.