adafruit/Adafruit_Blinka #798: AttributeError: module 'adafruit_platformdetect.constants.chips' has no attribute 'JH71x0'

anthonyisadulting, 2024-03-19:
Hello all! I am working with this library with a Jetson Nano, and I was wondering how to fix: 

AttributeError: module 'adafruit_platformdetect.constants.chips' has no attribute 'JH71x0'

I looked at the chips.py file and saw that there is an attribute there but the 'x' in that file is _capitalized_

Is that the reason why it is not reading properly? Thank you so much for any help, it is much appreciated.

-Ony

makermelissa, 2024-03-19:
Please make sure your version of Blinka has been updated. This was fixed on September 8, 2023 in https://github.com/adafruit/Adafruit_Blinka/commit/1530cf9cbfb008ff6a7e54e695dcb5a34f95d5a4#diff-9fca1fe7549121157e443209ee47d75044bc44e6c0297a9596f4eb147f089b07.

You can upgrade by running `pip install --upgrade adafruit-blinka`.

anthonyisadulting, 2024-03-20:
Just updated my adafruit blinka (used pip3 is that okay?) and I'm still receiving the same error message. D:

anthonyisadulting, 2024-03-21:
Ohhhh, yeah I'm using 3.6.9. I'll update my python when I get off of work today and I'll let you know!!!! Thank you so much for the fast responses Melissa, do you have any social media by chance? I appreciate your help so much !! 😭

makermelissa, 2024-03-21:
You're welcome. Social links are in my GitHub profile.

makermelissa, 2024-03-21:
Closing for now. We can always re-open.

makermelissa, 2024-03-21:
It looks like 3.8 was available, so I updated the guide for that.

anthonyisadulting, 2024-03-22:
Hi! Just wanted to make sure, does adafruit-blinka come with its own board library/module or would I have to install it on my own? I updated my python to 3.8 and now my VSCode isn't reading getting some modules.

makermelissa, 2024-03-22:
It sounds like you got a bit further. I'll try it out. I have the StarFive board as well and was the one who added it.

anthonyisadulting, 2024-03-22:
Oh, I'm using a Jetson Nano! I'm not sure if it's the same problem?

makermelissa, 2024-03-22:
Hmm, ok. I have that too and am working on it. Mine is not having the same issue.

makermelissa, 2024-03-22:
What is the code you are trying to run?

makermelissa, 2024-03-22:
Actually, I wonder if you are running an old version of PlatformDetect. Try running `pip install --upgrade Adafruit-PlatformDetect`.

makermelissa, 2024-03-22:
Try updating to `3.62.0`. 3.8.0 is a very old version. See: https://github.com/adafruit/Adafruit_Python_PlatformDetect/releases

makermelissa, 2024-03-22:
Also, why are you importing RPi.GPIO? That's for the Raspberry Pi.

anthonyisadulting, 2024-03-22:
3.62.0 or 3.6.2? And I'm importing RPi.GPIO because apparently it can be used for Jetson Nano as well? I'm unsure I have no lines in my code for that right now, I'm just using serial for motor control in my other functions (I've tried those and they worked before I implimented the stuff for neopixel)

makermelissa, 2024-03-22:
3.62.0
[image]

Yeah, I just learned that about RPi.GPIO while trying to add PWM support.

anthonyisadulting, 2024-03-22:
And just to confirm, this would be okay with python 3.8 right? Yeah, sorry I thought you meant python version!

anthonyisadulting, 2024-03-22:
Also I just updated my PlatformDetect to 3.62.0 and checked the version with pip3list and its the version you told me, unfortunately its still giving the same error :((

makermelissa, 2024-03-22:
It should be fine with 3.8. I just tested it. :) I'll try running your code and see if I run into the same thing.

makermelissa, 2024-03-22:
Based on the line numbers in your error message, I think you're still running an old version of Blinka. Please try running `python -m pip install --upgrade adafruit-blinka`.

anthonyisadulting, 2024-03-23:
Okay, that made some progress I think!

`Now I'm getting OSError: /dev/spidev0.0 does not exist`

I'm gonna look at the `sudo /opt/nvidia/jetson-io/jetson-io.py` again to make sure I didn't miss anything in those steps but if you have seen this issue before let me know!

anthonyisadulting, 2024-03-23:
Actually I just realized even though I enabled the spi1 through the script, nothing comes up with `ls /dev/spi*`

I'm gonna try and redo those changes, maybe I didn't reboot properly so it didn't write the file that the guide talks about.

anthonyisadulting, 2024-03-29:
Hi, I'm still having an issue with this. I currently have HDMI connected to the jetson nano, do you think that this is causing the problem?

makermelissa, 2024-06-04:
My understanding of the issue is it is caused by a combination of 2 things.
1. Blinka currently requires Python 3.7 or later and Jetson Nano comes with Python 3.6 by default (this can be upgraded to 3.8 by following [the guide](https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano/initial-setup#set-your-python-install-to-python-3-default-3039591).
2. There was a typo in the release back when Python 3.6 was supported, which is what is installed when Python 3.6 is the installed version.

The way to fix it is to first upgrade python and then update Blinka.

makermelissa, 2024-07-22:
Closing as this is a duplicate of #821, which has been resolved.

closed 2024-07-22 by makermelissa