adafruit/Adafruit_Blinka #813: Nanopi neo

Silvanskii, 2024-04-16:
root@OpenMPTCProuter:~/OLED_Stats# python3 psutilstats.py
Traceback (most recent call last):
  File "/root/OLED_Stats/psutilstats.py", line 9, in <module>
    import board
  File "/usr/lib/python3.11/site-packages/board.py", line 405, in <module>
    raise NotImplementedError(
NotImplementedError:
        Adafruit-PlatformDetect version 3.62.0 was unable to identify the board and/or
        microcontroller running the Linux platform. Please be sure you
        have the latest packages by running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

        If you are running the latest package, your board may not yet be supported. Please
        open a New Issue on GitHub at https://github.com/adafruit/Adafruit_Blinka/issues and
        select New Board Request.

nopnop2002, 2024-05-31:
This library supports nanopineno.
https://github.com/adafruit/Adafruit_Blinka/blob/main/src/board.py#L328

Are you using Armbian images?
https://www.armbian.com/nanopi-neo/

makermelissa, 2025-08-12:
This has been added: https://circuitpython.org/blinka/nanopi_neo_h3/