adafruit/Adafruit_Blinka #816: LED Channel as one of parameters for NeoPixel

ExcitePrime, 2024-04-19:
Hello, can you please add the LED Channel as one of the parameters when creating an instance of NeoPixel? This is so that we could use GPIO13 and GPIO19, which are on a different LED channel on a Raspberry Pi 4
[image]

dhalbert, 2024-04-19:
Transferring this issue to the Blinka repo. I don't understand what the issue is, but it seems to be a limitation in how to deal with the PWM channels.

makermelissa, 2024-04-19:
I'm going to reopen as it's a good suggestion, but I'll need to think about the best way to implement this without modifying the neopixel library.

makermelissa-ai-assistant, 2026-10-20:
Confirmed the requested single-strip GPIO13/PWM1 functionality on hardware with current `main` (`9b882eac59b97ad3633552721abeb369588851e3`). No constructor channel parameter or production-code change was needed.

Setup:

- Raspberry Pi 4 Model B, 64-bit Debian 13 (Trixie), kernel `6.18.50+rpt-rpi-v8`, Python 3.13.5.
- 30 RGB pixels, DIN on BCM GPIO13 (physical pin 33), 5 V power, common ground.
- `rpi_ws281x` 5.0.0, `adafruit-circuitpython-neopixel` 6.4.2, `adafruit-circuitpython-pixelbuf` 2.1.0.

The normal `neopixel.NeoPixel(board.D13, 30, ...)` path automatically selected native PWM channel 1, with GPIO13/count 30 and channel 0 unused. The user confirmed the expected dim red → green → blue → off sequence, repeated twice. Only the first pixel was illuminated; this was not an all-pixel or waveform-timing test. Cleanup returned GPIO13 to input mode.

The existing onboard PWM-audio driver and a temporary hardware-PWM overlay were removed for the test to avoid shared-peripheral conflicts, then restored. No boot configuration files were changed.

This support was added in [75033f1](https://github.com/adafruit/Adafruit_Blinka/commit/75033f1492f437db15d3e290d93ecd4cac55a6c7) on January 20, 2025. `_neopixel_detect_channel()` selects channel 1 for GPIO13/19, preserving the normal CircuitPython-compatible NeoPixel API.

GPIO19 uses the same existing channel-selection path but was not physically tested here. This does not claim simultaneous independent-strip support; Blinka's existing one-strip limitation remains. The original single-strip need appears resolved, so this issue can be closed without another implementation PR.

closed 2026-10-20 by makermelissa