adafruit/Adafruit_Blinka #821: Jetson Nano -- AttributeError: module 'adafruit_platformdetect.constants.chips' has no attribute 'JH71x0'

dorukcapraz, 2024-04-30:
Hello,

I recently bought a BMP280 sensor and I try to use it with the circuitpython_BMP280 library. I installed required packages but when I run the example code it is failing on the below line and giving the following error. Can anyone help me to fix this? I am using python 3.6.9, could it be the issue?

**The line which fails:**
i2c = board.I2C()

**Error Code:**
_Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/doruk/.local/lib/python3.6/site-packages/board.py", line 280, in I2C
    return busio.I2C(SCL, SDA)
  File "/home/doruk/.local/lib/python3.6/site-packages/busio.py", line 32, in __init__
    self.init(scl, sda, frequency)
  File "/home/doruk/.local/lib/python3.6/site-packages/busio.py", line 108, in init
    from microcontroller.pin import i2cPorts
  File "/home/doruk/.local/lib/python3.6/site-packages/microcontroller/__init__.py", line 65, in <module>
    elif chip_id == ap_chip.JH71x0:
AttributeError: module 'adafruit_platformdetect.constants.chips' has no attribute 'JH71x0'_

makermelissa, 2024-05-01:
First, make sure you have the latest PlatformDetect. I see you're also using Python 3.6. You can follow this guide to update it to 3.8, which will allow you to install the latest versions of Blinka/PlatformDetect: https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano/initial-setup#set-your-python-install-to-python-3-default-3039591

closed 2024-05-02 by makermelissa

dorukcapraz, 2024-05-04:
Than you it worked for me with the python3.8! I was trying to communicate with on board UART (Pin 8 & Pin 10). I had to activate below lines to make it work. Also I was trying to run the code in my house, I was not able to get the info from the sensor. When I put the sensor outside of my house, I was able to see the location.
By default this code comes with the example code
_uart = serial.Serial("/dev/ttyUSB0", baudrate=9600, timeout=10)_
_The part that I activated_
import serial
uart = serial.Serial("/dev/ttyTHS1", baudrate=9600, timeout=3000)