adafruit/Adafruit_Blinka #822: NotImplementedError:          Adafruit-PlatformDetect version 3.62.0 was unable to identify the board and/or         microcontroller running the Windows platform. Please be sure you         have the latest packages by running:         'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'          If you are running the latest package, your board may not yet be supported. Please         open a New Issue on GitHub at https://github.com/adafruit/Adafruit_Blinka/issues and         select New Board Request

mano-sys, 2024-05-02:


makermelissa, 2024-05-02:
Which board are you using?

alorcamo, 2024-05-03:
I get the exact same error (version 3.62.0). Running Windows 11 connected to a Adafruit FT232H. 

Any clues on how to fix it? I want to connect to a Qwiic sensor from a python script on the Windows 11 PC over the Adafruit FT232H. Following the [CircuitPython Libraries on any Computer with FT232H](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h)  Windows](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/windows) tutorial.

alorcamo, 2024-05-03:
Nevermind, solved. I realized too late that the VS Code Terminal is Powershell and I had used `set BLINKA_FT232H=1` instead of  `$env:BLINKA_FT232H=1`...I managed to oversee the big warning (:  ... just in case it helps someone.

makermelissa, 2024-05-03:
Awesome, thanks for reporting the fix for you.

closed 2024-05-03 by makermelissa