adafruit/Adafruit_Blinka #824: Add board: BOARD=orangepi3-lts

dgvigil, 2024-05-03:
I would like to request that the Orange Pi 3 LTS board be added.

```
orangepi@opi3lts-01:~$ cat /etc/orangepi-release 
# PLEASE DO NOT EDIT THIS FILE
BOARD=orangepi3-lts
BOARD_NAME="OPI 3 LTS"
BOARDFAMILY=sun50iw6
BUILD_REPOSITORY_URL=https://github.com/orangepi-xunlong/orangepi-build
BUILD_REPOSITORY_COMMIT=eb486d7-dirty
DISTRIBUTION_CODENAME=jammy
DISTRIBUTION_STATUS=supported
VERSION=3.0.8
LINUXFAMILY=sun50iw6
ARCH=arm64
IMAGE_TYPE=user-built
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
```

I saw that the Orange Pi 3 is support, but not the LTS. I did make sure to install the latest version as well.

```
orangepi@opi3lts-01:~$ pip3 install --upgrade adafruit-blinka adafruit-platformdetect
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: adafruit-blinka in ./.local/lib/python3.10/site-packages (8.39.1)
Requirement already satisfied: adafruit-platformdetect in ./.local/lib/python3.10/site-packages (3.62.0)
Requirement already satisfied: pyftdi>=0.40.0 in ./.local/lib/python3.10/site-packages (from adafruit-blinka) (0.55.4)
Requirement already satisfied: Adafruit-PureIO>=1.1.7 in ./.local/lib/python3.10/site-packages (from adafruit-blinka) (1.1.11)
Requirement already satisfied: adafruit-circuitpython-typing in ./.local/lib/python3.10/site-packages (from adafruit-blinka) (1.10.3)
Requirement already satisfied: pyserial>=3.0 in ./.local/lib/python3.10/site-packages (from pyftdi>=0.40.0->adafruit-blinka) (3.5)
Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in ./.local/lib/python3.10/site-packages (from pyftdi>=0.40.0->adafruit-blinka) (1.2.1)
Requirement already satisfied: typing-extensions~=4.0 in ./.local/lib/python3.10/site-packages (from adafruit-circuitpython-typing->adafruit-blinka) (4.11.0)
Requirement already satisfied: adafruit-circuitpython-requests in ./.local/lib/python3.10/site-packages (from adafruit-circuitpython-typing->adafruit-blinka) (3.2.8)
Requirement already satisfied: adafruit-circuitpython-busdevice in ./.local/lib/python3.10/site-packages (from adafruit-circuitpython-typing->adafruit-blinka) (5.2.9)
Requirement already satisfied: Adafruit-Circuitpython-ConnectionManager in ./.local/lib/python3.10/site-packages (from adafruit-circuitpython-requests->adafruit-circuitpython-typing->adafruit-blinka) (2.0.0)
orangepi@opi3lts-01:~$ sudo python3 bme280_simpletest.py
Traceback (most recent call last):
  File "/home/orangepi/bme280_simpletest.py", line 5, in <module>
    import board
  File "/usr/local/lib/python3.10/dist-packages/board.py", line 405, in <module>
    raise NotImplementedError(
NotImplementedError:
        Adafruit-PlatformDetect version 3.62.0 was unable to identify the board and/or
        microcontroller running the Linux platform. Please be sure you
        have the latest packages by running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

        If you are running the latest package, your board may not yet be supported. Please
        open a New Issue on GitHub at https://github.com/adafruit/Adafruit_Blinka/issues and
        select New Board Request.
```

nopnop2002, 2024-05-31:
This library supports orangepi3.
https://github.com/adafruit/Adafruit_Blinka/blob/main/src/board.py#L130

Are you using Armbian images?
https://www.armbian.com/orange-pi-3/

makermelissa, 2025-08-12:
Duplicate of https://github.com/adafruit/Adafruit_Blinka/issues/733

makermelissa, 2025-08-12:
Oops, not a duplicate.