adafruit/Adafruit_Blinka #838: Cant find the board

normjust2bad, 2024-06-04:
DEVICE ID 0403:6014 on Bus 000 Address 255 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x0403
 idProduct              : 0x6014
 bcdDevice              :  0x900 Device 9.0
 iManufacturer          :    0x1 FTDI
 iProduct               :    0x2 Single RS232-HS
 iSerialNumber          :    0x0
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 500 mA ==================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x20 (32 bytes)
   bNumInterfaces       :    0x1
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0
   bmAttributes         :   0x80 Bus Powered
   bMaxPower            :   0xfa (500 mA)
    INTERFACE 0: Vendor Specific ===========================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :   0xff Vendor Specific
     bInterfaceSubClass :   0xff
     bInterfaceProtocol :   0xff
     iInterface         :    0x2 Single RS232-HS
      ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
      ENDPOINT 0x2: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x2 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
None
Traceback (most recent call last):
  File "d:\Python\Robot le coin du jeu\test.py", line 17, in <module>
    import board
  File "D:\Python\.venv\Lib\site-packages\board.py", line 407, in <module>
    raise NotImplementedError(
NotImplementedError:
        Adafruit-PlatformDetect version 3.68.0 was unable to identify the board and/or
        microcontroller running the Windows platform. Please be sure you
        have the latest packages by running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

        If you are running the latest package, your board may not yet be supported. Please
        open a New Issue on GitHub at https://github.com/adafruit/Adafruit_Blinka/issues and
        select New Board Request.

PS D:\Python>

makermelissa, 2024-06-04:
Please provide which board it was unable to detect.

caternuson, 2024-06-04:
Looks like a FTDI FT232H based on USB VID/PID
```
DEVICE ID 0403:6014
```

Did all the post install checks pass?
https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/troubleshooting

makermelissa, 2024-06-04:
Thanks @caternuson

normjust2bad, 2024-06-04:
thank you for your help, i think i will return the amazon module and buy
another one
im trying to access a I2C bus with my windows portable PC,, some GPIO pin
would be great too

do you have a board to suggest ?

thank you very much
Normand Plante, Montreal

caternuson, 2024-06-05:
The FT232H:
https://www.adafruit.com/product/2264
and the MCP2221A are good options:
https://www.adafruit.com/product/4471

There's also u2if on an RP2040 board as an option:
https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-raspberry-pi-pico/overview

If you are having issues getting it to work per any of the Adafruit guides, you can post in the forums with details about what is not working, like which of the post install checks for the FT232H are failing, etc.
https://forums.adafruit.com/

makermelissa, 2024-07-22:
Closing for now as this appears to be a support issue and not a code issue.

closed 2024-07-22 by makermelissa