adafruit/Adafruit_Blinka #845: GT911 touch sensor support for Adafruit Blinka

leonard-voss, 2024-06-11:
Currently there is only this library available for the GT911 touch sensor:

[https://github.com/rgrizzell/CircuitPython_GT911](https://github.com/rgrizzell/CircuitPython_GT911)

I think that it only works on boards, on which native CircuitPython is running. Blinka-boards like the Raspberry Pi are not supported yet
(or at least not without major configuration effort).

It would be really nice to use a standard CircuitPython library for this.

makermelissa, 2024-06-11:
Blinka is actually designed to work with libraries like this. However, it is a little behind in the number of modules compared to CircuitPython. Is there a missing module that needs to be added to work with that library?

leonard-voss, 2024-06-12:
> Blinka is actually designed to work with libraries like this. However, it is a little behind in the number of modules compared to CircuitPython. Is there a missing module that needs to be added to work with that library?

Hey, thanks for replying. It seems like there are compatibility issues (with the Raspberry Pi) when using the library with Blinka. I described the whole thing in detail in this issue [here](https://github.com/rgrizzell/CircuitPython_GT911/issues/5).

makermelissa, 2024-06-12:
Ok, that makes more sense. So the pins need to support open drain. Are you using a Pi 5 or something else?

leonard-voss, 2024-06-12:
> Ok, that makes more sense. So the pins need to support open drain. Are you using a Pi 5 or something else?

I'm currently using a Raspberry Pi 4 Model B. But I think the Raspberry Pi does not support Open Drain mode in general or am I wrong?

leonard-voss, 2024-06-12:
OK. But then it all seems to support my suspicion. What is the best way to proceed now?

leonard-voss, 2024-06-18:
Alright, so I solved the issue (i had an wiring issue) and now the library works fine. Thanks for your support!

closed 2024-06-18 by leonard-voss