adafruit/Adafruit_Blinka #847: adafruit-circuitpython-motor fails with "RuntimeError: Cannot determine SOC peripheral base address" on Raspberry Pi 5

linucks, 2024-06-13:
### Board Name

Raspberry Pi 5

### Steps

Following the [instructions](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi) to install adafruit on a Raspberry Pi 5. I used the automated raspi-blinka.py script (see logs below). I then ran the  blinkatest.py file, changing board.D4 to board.D0 as I have the onewire interface enabled. The script runs successfully. I then tried to follow the example on [servo control](https://learn.adafruit.com/using-servos-with-circuitpython/high-level-servo-control) and the following script:

```
import board
import pwmio
pwm = pwmio.PWMOut(board.D0, frequency=50)
```

Fails with:
```
Traceback (most recent call last):
  File "/opt/fufarm_sensors/sensors/rpi_arduino_shield/foo.py", line 5, in <module>
    pwm = pwmio.PWMOut(board.D5, frequency=50)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/fufarm_sensors/sensors/rpi_arduino_shield/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pwmio/PWMOut.py", line 27, in __init__
    self._open(pin, duty_cycle, frequency, variable_frequency)
  File "/opt/fufarm_sensors/sensors/rpi_arduino_shield/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pwmio/PWMOut.py", line 40, in _open
    GPIO.setup(pin.id, GPIO.OUT)
RuntimeError: Cannot determine SOC peripheral base address
```

I saw that there was an issue #776 that was supposed to fix this, but the code from that PR is present and I'm still seeing the above error.

### Description

This is on a Raspberry Pi 5

Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

uname -a:
Linux jenspi 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

### Additional information

Logs from raspi-blinka.py:

```
This script configures your
Raspberry Pi and installs Blinka

RASPBERRY_PI_5 detected.

Updating System Packages
Blinka Hit:1 http://deb.debian.org/debian bookworm InRelease
Blinka Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Blinka Hit:3 http://archive.raspberrypi.com/debian bookworm InRelease
Blinka Hit:4 http://deb.debian.org/debian bookworm-updates InRelease
Blinka Hit:5 https://download.docker.com/linux/debian bookworm InRelease
Blinka Get:6 https://pkgs.tailscale.com/stable/raspbian bookworm InRelease
Blinka Fetched 6,578 B in 1s (9,899 B/s)
Reading package lists...
Blinka 
Blinka W: https://download.docker.com/linux/debian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Upgrading packages...
Blinka Reading package lists...
Blinka 
Blinka Building dependency tree...
Blinka 
Blinka Reading state information...
Blinka 
Blinka Calculating upgrade...
Blinka 
Blinka The following packages have been kept back:
Blinka lxplug-updater wf-panel-pi
Blinka 0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Making sure the required version of Python is installed
Enabling I2C
Enabling SPI
Enabling Serial
Enabling SSH
Enabling Camera
Disable raspi-config at Boot
Making sure Python 3 is the default
Making sure PIP and setuptools is installed
Blinka Reading package lists...
Blinka 
Blinka Building dependency tree...
Blinka Reading state information...
Blinka 
Blinka python3-pip is already the newest version (23.0.1+dfsg-1+rpt1).
python3-setuptools is already the newest version (66.1.1-1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Installing latest version of Blinka locally
Blinka Reading package lists...
Blinka 
Blinka Building dependency tree...
Blinka 
Blinka Reading state information...
Blinka 
Blinka i2c-tools is already the newest version (4.3-2+b3).
libgpiod-dev is already the newest version (1.6.3-1+b3).
python3-libgpiod is already the newest version (1.6.3-1+b3).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Blinka Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Blinka Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (0.7.1a4)
Blinka Collecting RPi.GPIO
Blinka Using cached RPi.GPIO-0.7.1-cp311-cp311-linux_aarch64.whl
Blinka Installing collected packages: RPi.GPIO
Blinka Attempting uninstall: RPi.GPIO
Blinka Found existing installation: RPi.GPIO 0.7.1a4
Blinka Not uninstalling rpi-gpio at /usr/lib/python3/dist-packages, outside environment /opt/fufarm_sensors/sensors/rpi_arduino_shield/env
Blinka Can't uninstall 'RPi.GPIO'. No files were found to uninstall.
Blinka Successfully installed RPi.GPIO-0.7.1
Blinka Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Blinka Collecting adafruit-blinka
Blinka Using cached https://www.piwheels.org/simple/adafruit-blinka/Adafruit_Blinka-8.41.2-py3-none-any.whl (348 kB)
Blinka Requirement already satisfied: Adafruit-PlatformDetect>=3.62.0 in ./env/lib/python3.11/site-packages (from adafruit-blinka) (3.69.0)
Blinka Collecting Adafruit-PureIO>=1.1.7
Blinka Using cached https://www.piwheels.org/simple/adafruit-pureio/Adafruit_PureIO-1.1.11-py3-none-any.whl (10 kB)
Blinka Collecting pyftdi>=0.40.0
Blinka Using cached https://www.piwheels.org/simple/pyftdi/pyftdi-0.55.4-py3-none-any.whl (145 kB)
Blinka Collecting adafruit-circuitpython-typing
Blinka Using cached https://www.piwheels.org/simple/adafruit-circuitpython-typing/adafruit_circuitpython_typing-1.10.3-py3-none-any.whl (11 kB)
Blinka Requirement already satisfied: RPi.GPIO in ./env/lib/python3.11/site-packages (from adafruit-blinka) (0.7.1)
Blinka Collecting rpi-ws281x>=4.0.0
Blinka Using cached rpi_ws281x-5.0.0-cp311-cp311-linux_aarch64.whl
Blinka Collecting sysv-ipc>=1.1.0
Blinka Using cached sysv_ipc-1.1.0-cp311-cp311-linux_aarch64.whl
Blinka Collecting pyusb!=1.2.0,>=1.0.0
Blinka Using cached https://www.piwheels.org/simple/pyusb/pyusb-1.2.1-py3-none-any.whl (58 kB)
Blinka Requirement already satisfied: pyserial>=3.0 in /usr/lib/python3/dist-packages (from pyftdi>=0.40.0->adafruit-blinka) (3.5)
Blinka Collecting adafruit-circuitpython-busdevice
Blinka Using cached https://www.piwheels.org/simple/adafruit-circuitpython-busdevice/adafruit_circuitpython_busdevice-5.2.9-py3-none-any.whl (7.5 kB)
Blinka Collecting adafruit-circuitpython-requests
Blinka Using cached https://www.piwheels.org/simple/adafruit-circuitpython-requests/adafruit_circuitpython_requests-4.0.0-py3-none-any.whl (10 kB)
Blinka Requirement already satisfied: typing-extensions~=4.0 in /usr/lib/python3/dist-packages (from adafruit-circuitpython-typing->adafruit-blinka) (4.4.0)
Blinka Collecting Adafruit-Circuitpython-ConnectionManager
Blinka Using cached https://www.piwheels.org/simple/adafruit-circuitpython-connectionmanager/adafruit_circuitpython_connectionmanager-3.1.0-py3-none-any.whl (7.8 kB)
Blinka Installing collected packages: sysv-ipc, rpi-ws281x, pyusb, Adafruit-PureIO, pyftdi, Adafruit-Circuitpython-ConnectionManager, adafruit-circuitpython-requests, adafruit-circuitpython-busdevice, adafruit-circuitpython-typing, adafruit-blinka
Blinka Successfully installed Adafruit-Circuitpython-ConnectionManager-3.1.0 Adafruit-PureIO-1.1.11 adafruit-blinka-8.41.2 adafruit-circuitpython-busdevice-5.2.9 adafruit-circuitpython-requests-4.0.0 adafruit-circuitpython-typing-1.10.3 pyftdi-0.55.4 pyusb-1.2.1 rpi-ws281x-5.0.0 sysv-ipc-1.1.0
DONE.

Settings take effect on next boot.

REBOOT NOW? [Y/n] 
```

makermelissa, 2024-06-13:
Thanks. I'll take a look in a couple of days after I wrap up my current project and can devote more time to this.

linucks, 2024-06-13:
Thanks @makermelissa if it looks like a genuine bug and there's anything I can do to help you get to the bottom of it, just let me know

brentru, 2024-06-17:
@makermelissa  I am experiencing the same error on Raspberry Pi 5 while testing CircuitPython's `PWMOut` module:

```
  File "/home/pi/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pwmio/PWMOut.py", line 27, in __init__
    self._open(pin, duty_cycle, frequency, variable_frequency)
  File "/home/pi/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pwmio/PWMOut.py", line 40, in _open
    GPIO.setup(pin.id, GPIO.OUT)
RuntimeError: Cannot determine SOC peripheral base address
```

makermelissa, 2024-06-17:
I'll take a look later today.

makermelissa, 2024-06-17:
Ok, I think I have a grasp on the issue. First, the Pi 5 is expecting `rpi-lgpio` to be installed. However, installing it results in a different error:
```
Traceback (most recent call last):
  File "/home/pi/pwmtest.py", line 4, in <module>
    pwm = pwmio.PWMOut(board.D5, frequency=50)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pwmio/PWMOut.py", line 27, in __init__
    self._open(pin, duty_cycle, frequency, variable_frequency)
  File "/home/pi/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pwmio/PWMOut.py", line 41, in _open
    self._pwmpin = GPIO.PWM(pin.id, freq)
                   ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/env/lib/python3.11/site-packages/RPi/GPIO/__init__.py", line 165, in __init__
    self._gpio = _to_gpio(channel)
                 ^^^^^^^^^^^^^^^^^
  File "/home/pi/env/lib/python3.11/site-packages/RPi/GPIO/__init__.py", line 360, in _to_gpio
    if not 0 <= channel < 54:
           ^^^^^^^^^^^^^^^^^
TypeError: '<=' not supported between instances of 'int' and 'tuple'
Exception ignored in: <function PWM.__del__ at 0x7fff7ffd3e20>
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.11/site-packages/RPi/GPIO/__init__.py", line 179, in __del__
    self.stop()
  File "/home/pi/env/lib/python3.11/site-packages/RPi/GPIO/__init__.py", line 202, in stop
    lgpio.tx_pwm(_chip, self._gpio, 0, 0)
                        ^^^^^^^^^^
AttributeError: 'PWM' object has no attribute '_gpio'
```

I believe this is because we are mixing and matching libraries (libgpiod for gpio and rpi-lgpio for pwm). Easy fix would be to use rpi-lgpio for GPIO too, but there doesn't seem to be a good way to tell it to use rpi-lgpio (it is a compatibility layer for RPi.GPIO) only for Pi 5. I'll see if I can figure out a way, but requirements.txt isn't as granular as setup.py. Then I'll update GPIO to use rpi-lgpio as well.

I may have painted myself into a corner a bit since libgpiod doesn't do PWM. So, fallback solution will be to make a copy of the PWMOut class and update both it and the GPIO use lgpio (without the compatibility layer).

makermelissa, 2024-06-18:
Ok good news is I was able to fix the setup to install rpi-lgpio and not RPi.GPIO for the Pi 5. That means moving over to using rpi-lgpio for the Pi 5, which should be about as simple as just having it use the Pi 4 stuff.

linucks, 2024-06-18:
Thanks for sorting that so quickly @makermelissa - I'll give that a try tomorrow.

closed 2024-06-19 by makermelissa

linucks, 2024-06-21:
@makermelissa I've upgraded everything but unfortunately, I'm still getting the same error when I run the blinkatest.py file:

```
Hello, blinka!
Traceback (most recent call last):
  File "/opt/fufarm_sensors/sensors/rpi_arduino_shield/blinkatest.py", line 8, in <module>
    pin = digitalio.DigitalInOut(board.D5)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/fufarm_sensors/sensors/rpi_arduino_shield/env/lib/python3.11/site-packages/digitalio.py", line 191, in __init__
    self.direction = Direction.INPUT
    ^^^^^^^^^^^^^^
  File "/opt/fufarm_sensors/sensors/rpi_arduino_shield/env/lib/python3.11/site-packages/digitalio.py", line 221, in direction
    self._pin.init(mode=Pin.IN)
  File "/opt/fufarm_sensors/sensors/rpi_arduino_shield/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pin.py", line 40, in init
    GPIO.setup(self.id, GPIO.IN)
RuntimeError: Cannot determine SOC peripheral base address
```
It looks like the changes you made are there. Is there any other reason why this might not be working for me? I'll paste in the output from the `raspi-blinka.py` script that I ran again in case it's helpful.

```
Ss script configures your
Raspberry Pi and installs Blinka

RASPBERRY_PI_5 detected.

Updating System Packages
Blinka Hit:1 http://archive.raspberrypi.com/debian bookworm InRelease
Blinka Hit:2 http://deb.debian.org/debian bookworm InRelease
Blinka Hit:3 http://deb.debian.org/debian-security bookworm-security InRelease
Blinka Hit:4 http://deb.debian.org/debian bookworm-updates InRelease
Blinka Hit:5 https://download.docker.com/linux/debian bookworm InRelease
Blinka Get:6 https://pkgs.tailscale.com/stable/raspbian bookworm InRelease
Blinka Fetched 6,578 B in 1s (10.6 kB/s)
Reading package lists...
Blinka 
Blinka W:
Blinka https://download.docker.com/linux/debian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Upgrading packages...
Blinka Reading package lists...
Blinka 
Blinka Building dependency tree...
Blinka 
Blinka Reading state information...
Blinka 
Blinka Calculating upgrade...
Blinka 
Blinka The following packages have been kept back:
Blinka libcamera-ipa libcamera-tools libpipewire-0.3-0 libpipewire-0.3-modules
  libspa-0.2-bluetooth libspa-0.2-modules lxplug-updater pipewire pipewire-bin
  pipewire-libcamera pipewire-pulse python3-libcamera rpicam-apps wf-panel-pi
Blinka 0 upgraded, 0 newly installed, 0 to remove and 14 not upgraded.
Making sure the required version of Python is installed
Enabling I2C
Enabling SPI
Enabling Serial
Enabling SSH
Enabling Camera
Disable raspi-config at Boot
Making sure Python 3 is the default
Making sure PIP and setuptools is installed
Blinka Reading package lists...
Blinka 
Blinka Building dependency tree...
Blinka 
Blinka Reading state information...
Blinka 
Blinka python3-pip is already the newest version (23.0.1+dfsg-1+rpt1).
python3-setuptools is already the newest version (66.1.1-1).
0 upgraded, 0 newly installed, 0 to remove and 14 not upgraded.
Installing latest version of Blinka locally
[... 29 lines cut ...]
Blinka Using cached https://www.piwheels.org/simple/pyftdi/pyftdi-0.55.4-py3-none-any.whl (145 kB)
Blinka Requirement already satisfied: numpy>=1.21.5 in /usr/lib/python3/dist-packages (from adafruit-blinka) (1.24.2)
Blinka Collecting adafruit-circuitpython-typing
Blinka Using cached https://www.piwheels.org/simple/adafruit-circuitpython-typing/adafruit_circuitpython_typing-1.10.3-py3-none-any.whl (11 kB)
Blinka Requirement already satisfied: RPi.GPIO in ./env/lib/python3.11/site-packages (from adafruit-blinka) (0.7.1)
Blinka Collecting rpi-ws281x>=4.0.0
Blinka Using cached rpi_ws281x-5.0.0-cp311-cp311-linux_aarch64.whl
Blinka Collecting sysv-ipc>=1.1.0
Blinka Using cached sysv_ipc-1.1.0-cp311-cp311-linux_aarch64.whl
Blinka Requirement already satisfied: pyserial in /usr/lib/python3/dist-packages (from binho-host-adapter>=0.1.6->adafruit-blinka) (3.5)
Blinka Collecting pyusb!=1.2.0,>=1.0.0
Blinka Using cached https://www.piwheels.org/simple/pyusb/pyusb-1.2.1-py3-none-any.whl (58 kB)
Blinka Collecting adafruit-circuitpython-busdevice
Blinka Using cached https://www.piwheels.org/simple/adafruit-circuitpython-busdevice/adafruit_circuitpython_busdevice-5.2.9-py3-none-any.whl (7.5 kB)
Blinka Collecting adafruit-circuitpython-requests
Blinka Using cached https://www.piwheels.org/simple/adafruit-circuitpython-requests/adafruit_circuitpython_requests-4.0.0-py3-none-any.whl (10 kB)
Blinka Requirement already satisfied: typing-extensions~=4.0 in /usr/lib/python3/dist-packages (from adafruit-circuitpython-typing->adafruit-blinka) (4.4.0)
Blinka Collecting Adafruit-Circuitpython-ConnectionManager
Blinka Using cached https://www.piwheels.org/simple/adafruit-circuitpython-connectionmanager/adafruit_circuitpython_connectionmanager-3.1.0-py3-none-any.whl (7.8 kB)
Blinka Installing collected packages: sysv-ipc, rpi-ws281x, pyusb, binho-host-adapter, Adafruit-PureIO, pyftdi, Adafruit-Circuitpython-ConnectionManager, adafruit-circuitpython-requests, adafruit-circuitpython-busdevice, adafruit-circuitpython-typing, adafruit-blinka
Blinka Successfully installed Adafruit-Circuitpython-ConnectionManager-3.1.0 Adafruit-PureIO-1.1.11 adafruit-blinka-8.45.0 adafruit-circuitpython-busdevice-5.2.9 adafruit-circuitpython-requests-4.0.0 adafruit-circuitpython-typing-1.10.3 binho-host-adapter-0.1.6 pyftdi-0.55.4 pyusb-1.2.1 rpi-ws281x-5.0.0 sysv-ipc-1.1.0
DONE.

settings take effect on next boot.ettings take effect on next boot.
```

makermelissa, 2024-06-21:
Try running `pip3 install rpi-lgpio`

linucks, 2024-06-26:
Sorry for the slow reply. My raspberry pi died and I had to get another one! I've just tried that and got the following:

```
 pip3 install rpi-lgpio
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting rpi-lgpio
  Downloading https://www.piwheels.org/simple/rpi-lgpio/rpi_lgpio-0.6-py3-none-any.whl (11 kB)
Requirement already satisfied: lgpio>=0.1.0.1 in /usr/lib/python3/dist-packages (from rpi-lgpio) (0.2.2.0)
Installing collected packages: rpi-lgpio
Successfully installed rpi-lgpio-0.6
```
Following that, it worked fine, so thank you for that!

Shouldn't the install process have installed the  rpi-lgpio library?

eMUQI, 2024-07-11:
Hello @makermelissa,

Today, I followed the installation tutorial again on a fresh Raspberry Pi setup, referring to the Automated Install section of [this guide](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi).

When I ran the following test code after completing the installation:

```python
import board
import digitalio
import busio

print("Hello, blinka!")

# Try to create a Digital input
pin = digitalio.DigitalInOut(board.D18)
print("Digital IO ok!")

# Try to create an I2C device
i2c = busio.I2C(board.SCL, board.SDA)
print("I2C ok!")

# Try to create an SPI device
spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
print("SPI ok!")

print("done!")
```

I encountered the `"RuntimeError: Cannot determine SOC peripheral base address"` error. Following the suggestion above, I installed `rpi-lgpio`, and everything returned to normal.

Perhaps it would be worth considering adding the installation of `rpi-lgpio` to the `raspi-blinka.py` script?

makermelissa, 2024-07-11:
> Perhaps it would be worth considering adding the installation of `rpi-lgpio` to the `raspi-blinka.py` script?

Yeah, that might be a good idea, however, it should already be automatic as it is part of the setup script: https://github.com/adafruit/Adafruit_Blinka/blob/main/setup.py#L40

I'll have to take a look more closely after I wrap up some CircuitPython Code Editor changes.

eMUQI, 2024-08-27:
Hi @makermelissa ,
This is really confusing. Today, I installed Blinka on a new Raspberry Pi 5 system again, but encountered the following error during testing:

```
Hello blinka!
Traceback (most recent call last):
  File "/home/pi3/btest.py", line 8, in <module>
    pin = digitalio.DigitalInOut(board.D4)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi3/env/lib/python3.11/site-packages/digitalio.py", line 193, in __init__
    self.direction = Direction.INPUT
    ^^^^^^^^^^^^^^
  File "/home/pi3/env/lib/python3.11/site-packages/digitalio.py", line 223, in direction
    self._pin.init(mode=Pin.IN)
  File "/home/pi3/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pin.py", line 40, in init
    GPIO.setup(self.id, GPIO.IN)
RuntimeError: Cannot determine SOC peripheral base address
```

I still followed the same installation steps and also installed `rpi-lgpio`. I'm not sure if I missed something or if there has been some change with the `blinka` library or with the Raspberry Pi itself. This makes me feel that Blinka is somewhat unstable; do you have any ideas? What can I do to help pinpoint the issue?

zhiyongwoo, 2024-08-28:
Hi @makermelissa 

I follow the steps in **Installing Blinka** to install Blinka, [Setup Virtual Environment](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi#setup-virtual-environment-3157129) and follow the [Automated Install](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi#automated-install-3081632).

Here is the console out put.
```Shell
This script configures your
Raspberry Pi and installs Blinka

RASPBERRY_PI_5 detected.

Updating System Packages
Blinka Hit:1 http://deb.debian.org/debian bookworm InRelease
Blinka Get:2 http://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Blinka Get:3 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Blinka Get:4 http://deb.debian.org/debian-security bookworm-security/main arm64 Packages [176 kB]
Blinka Get:5 http://deb.debian.org/debian-security bookworm-security/main armhf Packages [172 kB]
Blinka Get:6 http://deb.debian.org/debian-security bookworm-security/main Translation-en [107 kB]
Blinka Get:7 http://archive.raspberrypi.com/debian bookworm InRelease [39.0 kB]
Blinka Get:8 http://archive.raspberrypi.com/debian bookworm/main armhf Packages [454 kB]
Blinka Get:9 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages [452 kB]
Blinka Fetched 1,503 kB in 3s (451 kB/s)
Reading package lists...
Blinka
Upgrading packages...
Blinka Reading package lists...
Blinka
Blinka Building dependency tree...
Blinka
Blinka Reading state information...
Blinka
Blinka Calculating upgrade...
Blinka
Blinka The following packages will be upgraded:
Blinka libpython3.11 libpython3.11-dev libpython3.11-minimal libpython3.11-stdlib
Blinka python3.11 python3.11-dev python3.11-minimal python3.11-venv
Blinka 8 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 11.8 MB of archives.
After this operation, 35.8 kB disk space will be freed.
Get:1 http://deb.debian.org/debian-security bookworm-security/main arm64 python3.11-venv arm64 3.11.2-6+deb12u3 [5,896 B]
Blinka Get:2 http://deb.debian.org/debian-security bookworm-security/main arm64 python3.11-dev arm64 3.11.2-6+deb12u3 [616 kB]
Blinka Get:3 http://deb.debian.org/debian-security bookworm-security/main arm64 libpython3.11-dev arm64 3.11.2-6+deb12u3 [4,387 kB]
Blinka Get:4 http://deb.debian.org/debian-security bookworm-security/main arm64 libpython3.11 arm64 3.11.2-6+deb12u3 [1,840 kB]
Blinka Get:5 http://deb.debian.org/debian-security bookworm-security/main arm64 python3.11 arm64 3.11.2-6+deb12u3 [573 kB]
Blinka Get:6 http://deb.debian.org/debian-security bookworm-security/main arm64 libpython3.11-stdlib arm64 3.11.2-6+deb12u3 [1,747 kB]
Blinka Get:7 http://deb.debian.org/debian-security bookworm-security/main arm64 python3.11-minimal arm64 3.11.2-6+deb12u3 [1,857 kB]
Blinka Get:8 http://deb.debian.org/debian-security bookworm-security/main arm64 libpython3.11-minimal arm64 3.11.2-6+deb12u3 [808 kB]
Blinka apt-listchanges: Reading changelogs...
Blinka Fetched 11.8 MB in 1s (12.0 MB/s)
Blinka (Reading database ...
Blinka (Reading database ...
Blinka 5%
(Reading database ... 10%
(Reading database ... 15%
(Reading database ... 20%
Blinka (Reading database ...
Blinka 25%
(Reading database ...
Blinka 30%
(Reading database ...
Blinka 35%
(Reading database ... 40%
(Reading database ...
Blinka 45%
(Reading database ... 50%
Blinka (Reading database ... 55%
Blinka (Reading database ...
Blinka 60%
Blinka (Reading database ...
Blinka 65%
[... 238 lines cut ...]

Then what I did is to install rpi-lgpio
```Shell
(env) amps_pi5@raspberrypi:~ $ pip3 install rpi-lgpio
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting rpi-lgpio
  Downloading https://www.piwheels.org/simple/rpi-lgpio/rpi_lgpio-0.6-py3-none-any.whl (11 kB)
Requirement already satisfied: lgpio>=0.1.0.1 in /usr/lib/python3/dist-packages (from rpi-lgpio) (0.2.2.0)
Installing collected packages: rpi-lgpio
Successfully installed rpi-lgpio-0.6
```

and try again on the blinkatest.py
```Shell
(env) amps_pi5@raspberrypi:~ $ python3 blinkatest.py
Hello, blinka!
Digital IO ok!
I2C ok!
SPI ok!
done!
```

@makermelissa Can we have the RPi.GPIO completely uninstall?

Many thanks.

makermelissa, 2024-08-28:
Yeah, with the way I thought I had it, RPi.GPIO should not have been installed and rpi-lgpio should have been installed instead.

laksirisamitha, 2024-09-10:
_check(lgpio.gpio_claim_input(_chip, gpio, {
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3/dist-packages/lgpio.py", line 755, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))
**lgpio.error: 'GPIO busy'**

I followed every step mentioned above correctly but finally got this error ? could you please help me on this ?

laksirisamitha, 2024-09-10:
_check(lgpio.gpio_claim_input(_chip, gpio, { ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/lib/python3/dist-packages/lgpio.py", line 755, in gpio_claim_input return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio)) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/lib/python3/dist-packages/lgpio.py", line 458, in _u2i raise error(error_text(v)) **lgpio.error: 'GPIO busy'**
> 
> I followed every step mentioned above correctly but finally got this error ? could you please help me on this ?

**I have enabled 1-Wire , That caused the issue. Thank you !**

PR #856 (2024-06-19): Fix PWM for the Pi 5
Fixes #847. This fixes PWM for the Pi 5 by using rpi-lgpio for the GPIO instead of libgpiod. To accomplish this, I needed it to install rpi-lgpio for the Pi 5 and RPi.GPIO for the Pi 4, so requirements.txt needed to run setup.py, which is able to determine the CPU model.