adafruit/Adafruit_Blinka #849: Adafruit-Adafruit-PlatformDetect version 3.69.0 was unable to identify the board

Hurricane-k, 2024-06-14:
Hello Everyone

`
NotImplementedError: 
Adafruit-PlatformDetect version 3.69.0 was unable to identify the board and/or microcontroller running the Windows platform. Please be sure you have the latest packages by running: pip3 install --upgrade adafruit-blinka adafruit-platformdetect.
`
`
If you are running the latest package, your board may not yet be supported. Please open a New Issue on GitHub at https://github.com/adafruit/Adafruit_Blinka/issues and select New Board Request.
`

Best Regards.

LPonGit, 2024-06-16:
Looks to me that you error has been caused by the import board command?
Please post the full error output, it should be more like this **File "/Users/lee/Documents/My Scripts/countup.py", line 5, in
import board**
File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/board.py", line 425, in
raise NotImplementedError(
NotImplementedError:
Adafruit-PlatformDetect version 3.69.0 was unable to identify the board and/or
microcontroller running the Darwin platform. Please be sure you
have the latest packages by running:
'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

What boards do you have connected ?
What version of Python are you using ?
run pip3 list and post the output

TommyWei-polyu, 2024-06-16:
I have the same issue.

> Traceback (most recent call last): Cell In[1], line 1, import board, File ~\AppData\Local\anaconda3\Lib\site-packages\board.py:425,raise NotImplementedError(
NotImplementedError: Adafruit-PlatformDetect version 3.69.0 was unable to identify the board and/or         microcontroller running the Windows platform. Please be sure you have the latest packages by running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect' 
        If you are running the latest package, your board may not yet be supported. Please
        open a New Issue on GitHub at https://github.com/adafruit/Adafruit_Blinka/issues and
        select New Board Request.

I am connecting to Adafruit AS731 [https://learn.adafruit.com/adafruit-as7341-10-channel-light-color-sensor-breakout/overview](url)

Python version is 3.11.7

pip3 list output:

> Package                                  Version
---------------------------------------- ------------
Adafruit-Blinka                          8.43.0
adafruit-circuitpython-as7341            1.2.18
adafruit-circuitpython-busdevice         5.2.9
adafruit-circuitpython-connectionmanager 3.1.0
adafruit-circuitpython-register          1.9.18
adafruit-circuitpython-requests          4.0.0
adafruit-circuitpython-typing            1.10.3
Adafruit-PlatformDetect                  3.69.0
Adafruit-PureIO                          1.1.11
aiobotocore                              2.7.0
aiohttp                                  3.9.3
aioitertools                             0.7.1
aiosignal                                1.2.0
alabaster                                0.7.12
altair                                   5.0.1
anaconda-anon-usage                      0.4.3
anaconda-catalogs                        0.2.0
anaconda-client                          1.12.3
anaconda-cloud-auth                      0.1.4
anaconda-navigator                       2.5.2
anaconda-project                         0.11.1
anyio                                    4.2.0
appdirs                                  1.4.4
archspec                                 0.2.1
argon2-cffi                              21.3.0
argon2-cffi-bindings                     21.2.0
arrow                                    1.2.3
astroid                                  2.14.2
astropy                                  5.3.4
asttokens                                2.0.5
async-lru                                2.0.4
atomicwrites                             1.4.0
attrs                                    23.1.0
Automat                                  20.2.0
autopep8                                 1.6.0
Babel                                    2.11.0
backports.functools-lru-cache            1.6.4
backports.tempfile                       1.0
backports.weakref                        1.0.post1
bcrypt                                   3.2.0
beautifulsoup4                           4.12.2
binaryornot                              0.4.4
black                                    23.11.0
bleach                                   4.1.0
blinker                                  1.6.2
bokeh                                    3.3.4
boltons                                  23.0.0
botocore                                 1.31.64
Bottleneck                               1.3.7
Brotli                                   1.0.9
cachetools                               4.2.2
certifi                                  2024.2.2
cffi                                     1.16.0
[... 302 lines cut ...]
uc-micro-py                              1.0.1
ujson                                    5.4.0
Unidecode                                1.2.0
urllib3                                  2.0.7
validators                               0.18.2
w3lib                                    2.1.2
watchdog                                 2.1.6
wcwidth                                  0.2.5
webencodings                             0.5.1
websocket-client                         0.58.0
Werkzeug                                 2.2.3
whatthepatch                             1.0.2
wheel                                    0.41.2
widgetsnbextension                       3.5.2
win-inet-pton                            1.1.0
wrapt                                    1.14.1
xarray                                   2023.6.0
xlwings                                  0.29.1
xyzservices                              2022.9.0
yapf                                     0.31.0
yarl                                     1.9.3
zict                                     3.0.0
zipp                                     3.17.0
zope.interface                           5.4.0
zstandard                                0.19.0

LPonGit, 2024-06-16:
Ghee you have a lot of Python experience judging by the installation of modules / libraries

OK, so Adafruit-Blinka and adafruit-circuitpython-as7341 1.2.18 are installed 👍

_**Point 1**_ I have noticed you don't have hidapi installed ? You will need hidapi.

How are you setting the Blinka Environment Variable within your Python script ?
I was using an Adafruit_ht16K33 4x 7 Segment
S0 I added the following lines to my script to **set the Environment Variable**

_**Point 2**_ You will need the same in your script
```
import os
os.environ["BLINKA_MCP2221"] = "1"

```
Let me know how you get on.

All the best 
Lee

TommyWei-polyu, 2024-06-16:
Point 1: I did install hidapi, which took me a while to figure it out. But it still does not work. 
Point 2: I set environment variable through windows. Does it work the same as through the script? 

The error I got was something related to "MCP2221" was not connected. I guess it was because I was using Arduino to connect the Adafruit board. I just ordered a "Adafruit Feather M4", which may solve the problem?

LPonGit, 2024-06-16:
I was setting the Evironment Variable from the terminal then launching my script from Python Launcher, this was giving me the error, but it was resolved instantly when I put 
```
import os
os.environ["BLINKA_MCP2221"] = "1"
```
 in my script

Hurricane-k, 2024-06-17:
> Looks to me that you error has been caused by the import board command? Please post the full error output, it should be more like this **File "/Users/lee/Documents/My Scripts/countup.py", line 5, in [...]

i have post the full error, that happens when i run `import board`. and i try this on Linux, it doesnt work too but it caused by different things. so i almost definitely confirm that is nothing to do with the windows platform

Hurricane-k, 2024-06-17:
> Looks to me that you error has been caused by the import board command? Please post the full error output, it should be more like this **File "/Users/lee/Documents/My Scripts/countup.py", line 5, in [...]

encounter this error when i make sure the board is connected, and can be recoginzed by my pc and my python is 3.7.4. and 3.69.0 is the latest version of this package. i mean detect platform

makermelissa, 2024-06-17:
@Hurricane-k which board do you have that was not detected?

Hurricane-k, 2024-06-18:
> @Hurricane-k which board do you have that was not detected?

the board is Arduino UNO

Hurricane-k, 2024-06-18:
> @Hurricane-k which board do you have that was not detected?

Do you mind if i ask you another thing related to it. 

When I try to use this board in Ubuntu 24.04, and in `python==3.9`.
When run `import board`, the result does not follow my expectation. `board.id_board='GENERIC_LINUX_PC'` means nothing more will be imported. Why does that happen.
Just cant run this Arduino board in Linux

makermelissa, 2024-06-18:
Ok, Arduino Uno isn't supported and likely won't be. Blinka is intended to run in 3 different ways:
1. It runs on boards that are able to host linux such as the Raspberry Pi
2. It runs on a few different boards on top of MicroPython such as the Raspberry Pi Pico
3. It runs on the computer to connect with the FT232H, MCP2221a, or RP2040-based boards with the UF2 firmware.

I'm fairly certain that the Arduino Uno doesn't fall under any of these categories. Even if MicroPython supports it, it likely doesn't have enough Flash to add Blinka as well.

> When I try to use this board in Ubuntu 24.04, and in `python==3.9`. When run `import board`, the result does not follow my expectation. `board.id_board='GENERIC_LINUX_PC'` means nothing more will be imported. Why does that happen. Just cant run this Arduino board in Linux

If I recall, it was to allow it to pass the GitHub Actions checks since it doesn't have actual hardware to test with.

Hurricane-k, 2024-06-19:
> Ok, Arduino Uno isn't supported and likely won't be. Blinka is intended to run in 3 different ways:
>
> 1. It runs on boards that are able to host linux such as the Raspberry Pi
> 2. It runs on a few [...]

thank you, got it. so being through Arduino IDE is the only way to upload code to Arduino Uno, is it right?

makermelissa, 2024-06-19:
> thank you, got it. so being through Arduino IDE is the only way to upload code to Arduino Uno, is it right?

I think you can also use VSCode with the PlatformIO extension or put Snek lang (similar to python) on it as well.

closed 2024-06-19 by makermelissa