adafruit/Adafruit_Blinka #865: It is no longer possible to install from 8.44.0.

nopnop2002, 2024-06-22:
### Board Name

Luckfox Pico Pro

### Steps

I was able to install 8.43.0 on my board.

```
$ pip list
Package                                  Version
---------------------------------------- -------
Adafruit-Blinka                          8.43.0
adafruit-circuitpython-bmp280            3.3.2
adafruit-circuitpython-busdevice         5.2.9
adafruit-circuitpython-connectionmanager 3.1.0
adafruit-circuitpython-register          1.9.18
adafruit-circuitpython-requests          4.0.0
adafruit-circuitpython-ticks             1.0.13
adafruit-circuitpython-typing            1.10.3
adafruit-circuitpython-wiznet5k          7.1.2
Adafruit-PlatformDetect                  3.69.0
Adafruit-PureIO                          1.1.11
```

From 8.44.0, the installation method has changed from requirements.txt to serum.py, and it can no longer be installed.

```
$ python3 -m pip install Adafruit-Blinka==8.44.0
Defaulting to user installation because normal site-packages is not writeable
Collecting Adafruit-Blinka==8.44.0
  Downloading Adafruit_Blinka-8.44.0-py3-none-any.whl.metadata (5.4 kB)
Collecting Adafruit-PlatformDetect>=3.70.1 (from Adafruit-Blinka==8.44.0)
  Using cached Adafruit_PlatformDetect-3.71.0-py3-none-any.whl.metadata (3.8 kB)
Requirement already satisfied: Adafruit-PureIO>=1.1.7 in ./.local/lib/python3.10/site-packages (from Adafruit-Blinka==8.44.0) (1.1.11)
Collecting binho-host-adapter>=0.1.6 (from Adafruit-Blinka==8.44.0)
  Using cached binho_host_adapter-0.1.6-py3-none-any.whl.metadata (2.0 kB)
Requirement already satisfied: pyftdi>=0.40.0 in ./.local/lib/python3.10/site-packages (from Adafruit-Blinka==8.44.0) (0.55.4)
Collecting numpy>=1.21.5 (from Adafruit-Blinka==8.44.0)
  Using cached numpy-2.0.0.tar.gz (18.3 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... error
  error: subprocess-exited-with-error

  × pip subprocess to install backend dependencies did not run successfully.
  x exit code: 1
  mq> [139 lines of output]
      Collecting patchelf>=0.11.0
        Using cached patchelf-0.17.2.1-py2.py3-none-manylinux_2_17_armv7l.manylinux2014_armv7l.musllinux_1_1_armv7l.whl.metadata (3.3 kB)
      Collecting ninja>=1.8.2
        Using cached ninja-1.11.1.1.tar.gz (132 kB)
        Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'done'
      Using cached patchelf-0.17.2.1-py2.py3-none-manylinux_2_17_armv7l.manylinux2014_armv7l.musllinux_1_1_armv7l.whl (381 kB)
      Building wheels for collected packages: ninja
        Building wheel for ninja (pyproject.toml): started
        Building wheel for ninja (pyproject.toml): finished with status 'error'
        error: subprocess-exited-with-error

        × Building wheel for ninja (pyproject.toml) did not run successfully.
        x exit code: 1
        mq> [114 lines of output]
            /tmp/pip-build-env-ny7bdnlh/overlay/local/lib/python3.10/dist-packages/setuptools_scm/git.py:312: UserWarning: git archive did not support describe output
              warnings.warn("git archive did not support describe output")
            /tmp/pip-build-env-ny7bdnlh/overlay/local/lib/python3.10/dist-packages/setuptools_scm/git.py:331: UserWarning: unprocessed git archival found (no export subst applied)
[... 95 lines cut ...]

              https://github.com/dockcross/dockcross#readme

            For more details, please refer to scikit-build documentation:

              http://scikit-build.readthedocs.io/en/latest/generators.html#linux

                            ********************************************************************************
            [end of output]

        note: This error originates from a subprocess, and is likely not a problem with pip.
        ERROR: Failed building wheel for ninja
      Failed to build ninja
      ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (ninja)
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install backend dependencies did not run successfully.
x exit code: 1
mq> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

nopnop2002, 2024-06-26:
```
            -- Detecting CXX compiler ABI info
            -- Detecting CXX compiler ABI info - failed
            -- Check for working CXX compiler: /usr/bin/c++
            -- Check for working CXX compiler: /usr/bin/c++ - broken
```

c++ was not installed correctly.
I solved it by reinstalling c++.

closed 2024-06-26 by nopnop2002

makermelissa, 2024-07-01:
Hmm, it didn't used to need to be installed. I think this may be a result of #863.

nopnop2002, 2024-07-01:
Starting from 8.45.0, numpy is installed at the same time, making installation very slow.

I don't understand the need to install numpy at the same time.

```
Collecting numpy>=1.21.5 (from Adafruit-Blinka)
```