adafruit/Adafruit_Blinka #870: Add New Board adafruit FT232H Breakout for adafruit-blinka

jungtom, 2024-07-18:
Unfortunately my Adafruit FT232H Breakout - General Purpose USB to GPIO+SPI+I2C [ADA2264] is not supported by current adafruit-blinka. It would be great if I could add the board to adafruit-blinka.

********************************

Blinke Version:
pip3 install --upgrade adafruit-blinka adafruit-platformdetect
Requirement already satisfied: adafruit-blinka in c:\apps\python310\lib\site-packages (8.46.0)
Requirement already satisfied: adafruit-platformdetect in c:\apps\python310\lib\site-packages (3.72.0)
Requirement already satisfied: Adafruit-PureIO>=1.1.7 in c:\apps\python310\lib\site-packages (from adafruit-blinka) (1.1.11)
Requirement already satisfied: binho-host-adapter>=0.1.6 in c:\apps\python310\lib\site-packages (from adafruit-blinka) (0.1.6)
Requirement already satisfied: pyftdi>=0.40.0 in c:\apps\python310\lib\site-packages (from adafruit-blinka) (0.55.4)
Requirement already satisfied: numpy>=1.21.5 in c:\apps\python310\lib\site-packages (from adafruit-blinka) (2.0.0)
Requirement already satisfied: adafruit-circuitpython-typing in c:\apps\python310\lib\site-packages (from adafruit-blinka) (1.10.3)
Requirement already satisfied: pyserial in c:\apps\python310\lib\site-packages (from binho-host-adapter>=0.1.6->adafruit-blinka) (3.5)
Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in c:\apps\python310\lib\site-packages (from pyftdi>=0.40.0->adafruit-blinka) (1.2.1)
Requirement already satisfied: adafruit-circuitpython-busdevice in c:\apps\python310\lib\site-packages (from adafruit-circuitpython-typing->adafruit-blinka) (5.2.9)
Requirement already satisfied: adafruit-circuitpython-requests in c:\apps\python310\lib\site-packages (from adafruit-circuitpython-typing->adafruit-blinka) (4.1.4)
Requirement already satisfied: typing-extensions~=4.0 in c:\apps\python310\lib\site-packages (from adafruit-circuitpython-typing->adafruit-blinka) (4.12.2)
Requirement already satisfied: Adafruit-Circuitpython-ConnectionManager in c:\apps\python310\lib\site-packages (from adafruit-circuitpython-requests->adafruit-circuitpython-typing->adafruit-blinka) (3.1.1)		

********************************
Board Info:
IC: FTDI DLJANC1 FT232HQ 2224-C

DEVICE ID 0403:6014 on Bus 000 Address 255 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x0403
 idProduct              : 0x6014
 bcdDevice              :  0x900 Device 9.0
 iManufacturer          :    0x1 ￿￿￿￿￿￿￿￿
 iProduct               :    0x2 ￿￿￿￿￿￿
 iSerialNumber          :    0x3 ￿￿￿￿￿￿￿￿
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 100 mA ==================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x20 (32 bytes)
   bNumInterfaces       :    0x1
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0 
   bmAttributes         :   0x80 Bus Powered
   bMaxPower            :   0x32 (100 mA)
    INTERFACE 0: Vendor Specific ===========================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :   0xff Vendor Specific
     bInterfaceSubClass :   0xff
     bInterfaceProtocol :   0xff
     iInterface         :    0x2 ￿￿￿￿￿￿
      ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
      ENDPOINT 0x2: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x2 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0

I hope the information is sufficient. 
Thank you for your support.
Greetings from Germany

makermelissa, 2024-07-18:
This has already been added. There is a learn guide available at https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h. If you need further support, you can either use [Discord](https://adafru.it/discord) or [the Adafruit forums](https://forums.adafruit.com/).

closed 2024-07-18 by makermelissa

jungtom, 2024-07-21:
Hello Melissa,

 

Thank you very much for your reply.
I have one more question.
When I run my script in the command prompt, it works fine. But normally I use PyCharm. But the script does not work here.

The error message in PyCharm is

 

Traceback (most recent call last):
  File "C:\_Projekt\workspace\py.projekt\Sandbox\Adafruit_GPIO\test_LED.py", line 2, in <module>
    import board
  File "C:\apps\Python310\lib\site-packages\board.py", line 433, in <module>
    raise NotImplementedError(
NotImplementedError:
        Adafruit-PlatformDetect version 3.72.1 was unable to identify the board and/or
        microcontroller running the Windows platform. Please be sure you
        have the latest packages by running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

        If you are running the latest package, your board may not yet be supported. Please
        open a New Issue on GitHub at https://github.com/adafruit/Adafruit_Blinka/issues and
        select New Board Request.

 

*****************

The script is very simple

import time
import board
import digitalio

led_aa = digitalio.DigitalInOut(board.C0)
led_aa.direction = digitalio.Direction.OUTPUT
led_bb = digitalio.DigitalInOut(board.D7)
led_bb.direction = digitalio.Direction.OUTPUT

while True:
    led_aa.value = True
    led_bb.value = False
    time.sleep(0.5)
    led_aa.value = False
    led_bb.value = True
    time.sleep(0.5)

 

I have performed the upgrade. The following libraries are installed.

 

Adafruit-Blinka                          8.46.0
adafruit-circuitpython-busdevice         5.2.9
adafruit-circuitpython-connectionmanager 3.1.1
adafruit-circuitpython-mcp3xxx           1.4.17
adafruit-circuitpython-requests          4.1.4
adafruit-circuitpython-typing            1.10.3
Adafruit_GPIO                            1.0.3
Adafruit-PlatformDetect                  3.72.1
Adafruit-PureIO                          1.1.11
adafruit-python-shell                    1.8.1

 

Do you have any idea where the problem might lie under PyCharm?

 

Many thanks for your help.

Thomas

Gesendet: Donnerstag, 18. Juli 2024 um 18:31 Uhr

makermelissa, 2024-07-22:
It's detected using environment variables. Perhaps PyCharm is hiding the environment variables from the library. Unfortunately, I haven't used PyCharm much and am not very familiar with it.