adafruit/Adafruit_Blinka #874: RPi GPIO is not support for Pi 5 any more, why merge pi5 with pi4 under board?

cavonlee, 2024-07-25:
### Board Name

Raspberry Pi 5

### Steps

1. Install Home assistant
2. create a custom addon
3. install adafruit blinka
4. open python, install board.

### Description

It's a wired issue. We are creating a addon for home assistant on Pi 5. And the addon  gets error from RPi.GPIO This module can only be run on a Raspberry Pi!

It use to work on 8.39.2. So we look inside the source code. And found differencies from latest 8.46.1. 
Pi5 now sharing the same board import, and bcm2712 is removed. That's wired, why is that? Is it a bug or a intend?

makermelissa, 2024-07-25:
RPi.GPIO does not support the Pi 5 anymore. A replacement library rpi-lgpio is used as a drop-in replacement for RPi.GPIO, so once that is installed (making sure RPi.GPIO is not installed), it should work fine.

makermelissa, 2024-07-25:
For a while, we were using libgpiod (which does not support PWM) for GPIO and rpi-lgpio for PWM. However, something changed the libraries and PWM was no longer working while using libgpiod, so rather than drop PWM support, we moved over to rpi-lgpio for GPIO as well.

makermelissa, 2024-08-10:
Closing as an explanation was provided.

closed 2024-08-10 by makermelissa