adafruit/Adafruit_Blinka #876: Maker Pi RP 2040 not supported.

GitbyteMaster, 2024-08-02:
I'd noticed that my board, a Maker Pi RP 2040 is not being recognized by Adafruit-PlatformDetect. I'm working on a Mac laptop, which registers as 'Darwin'. PlatformDetect is on version 3.72.1, which I found to be the latest version I could install. Here is my error. Python is on 3.11.4.
```python
raceback (most recent call last):
  File "/Volumes/CIRCUITPY/WPILib/WPILib.py", line 1, in <module> 
    import board as _board
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/board.py", line 433, in <module>
    raise NotImplementedError(
NotImplementedError: 
        Adafruit-PlatformDetect version 3.72.1 was unable to identify the board and/or
        microcontroller running the Darwin platform. Please be sure you
        have the latest packages by running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

        If you are running the latest package, your board may not yet be supported. Please
        open a New Issue on GitHub at https://github.com/adafruit/Adafruit_Blinka/issues and
        select New Board Request. # What led me here.
```

anecdata, 2024-08-02:
Blinka is not needed. You can run CircuitPython directly:

GitbyteMaster, 2024-08-02:
Oh. Alright.

closed 2024-08-02 by GitbyteMaster

anecdata, 2024-08-02:
Additional resource here:
https://github.com/CytronTechnologies/MAKER-PI-RP2040

GitbyteMaster, 2024-08-02:
Alright, thank you.