adafruit/Adafruit_Blinka #885: unable to install blinka on Luckfox Pico Mini

TheRegularDX, 2024-08-13:
### Board Name

Luckfox Pico Mini B

### Steps

run `pip3 install Adafruit-Blinka`
output:
`Defaulting to user installation because normal site-packages is not writeable
Collecting Adafruit-Blinka
  Using cached Adafruit_Blinka-8.47.0-py3-none-any.whl (356 kB)
Collecting Adafruit-PlatformDetect>=3.70.1
  Using cached Adafruit_PlatformDetect-3.73.0-py3-none-any.whl (24 kB)
Collecting Adafruit-PureIO>=1.1.7
  Using cached Adafruit_PureIO-1.1.11-py3-none-any.whl (10 kB)
Collecting pyftdi>=0.40.0
  Using cached pyftdi-0.55.4-py3-none-any.whl (145 kB)
Collecting sysv-ipc>=1.1.0
  Using cached sysv_ipc-1.1.0.tar.gz (99 kB)
  Preparing metadata (setup.py) ... done
Collecting binho-host-adapter>=0.1.6
  Using cached binho_host_adapter-0.1.6-py3-none-any.whl (10 kB)
Collecting adafruit-circuitpython-typing
  Using cached adafruit_circuitpython_typing-1.11.0-py3-none-any.whl (11 kB)
Collecting pyserial
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting pyusb!=1.2.0,>=1.0.0
  Using cached pyusb-1.2.1-py3-none-any.whl (58 kB)
Collecting typing-extensions~=4.0
  Using cached typing_extensions-4.12.2-py3-none-any.whl (37 kB)
Collecting adafruit-circuitpython-busdevice
  Using cached adafruit_circuitpython_busdevice-5.2.9-py3-none-any.whl (7.5 kB)
Collecting adafruit-circuitpython-requests
  Using cached adafruit_circuitpython_requests-4.1.6-py3-none-any.whl (10 kB)
Collecting Adafruit-Circuitpython-ConnectionManager
  Using cached adafruit_circuitpython_connectionmanager-3.1.1-py3-none-any.whl (7.8 kB)
Building wheels for collected packages: sysv-ipc
  Building wheel for sysv-ipc (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [24 lines of output]
      ******************************************************************************
      * Setup can't determine the value of PAGE_SIZE on your system, so it will
      * default to 4096 which may not be correct.
      *
      * Please report this message and your operating system info to the package
      * maintainer listed in the README file.
      ******************************************************************************
      running bdist_wheel
      running build
      running build_ext
      building 'sysv_ipc' extension
      creating build
      creating build/temp.linux-armv7l-3.10
      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.10 -c common.c -o build/temp.linux-armv7l-3.10/common.o
      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.10 -c memory.c -o build/temp.linux-armv7l-3.10/memory.o
      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.10 -c mq.c -o build/temp.linux-armv7l-3.10/mq.o
      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.10 -c semaphore.c -o build/temp.linux-armv7l-3.10/semaphore.o
      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.10 -c sysv_ipc_module.c -o build/temp.linux-armv7l-3.10/sysv_ipc_module.o
      creating build/lib.linux-armv7l-3.10
      arm-linux-gnueabihf-gcc -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -g -fwrapv -O2 -Wl,-Bsymbolic-functions -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.10/common.o build/temp.linux-armv7l-3.10/memory.o build/temp.linux-armv7l-3.10/mq.o build/temp.linux-armv7l-3.10/semaphore.o build/temp.linux-armv7l-3.10/sysv_ipc_module.o -o build/lib.linux-armv7l-3.10/sysv_ipc.cpython-310-arm-linux-gnueabihf.so
      /usr/bin/ld: cannot find -lgcc: No such file or directory
      /usr/bin/ld: cannot find -lgcc: No such file or directory
      collect2: error: ld returned 1 exit status
      error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
[... 29 lines cut ...]
      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.10 -c sysv_ipc_module.c -o build/temp.linux-armv7l-3.10/sysv_ipc_module.o
      creating build/lib.linux-armv7l-3.10
      arm-linux-gnueabihf-gcc -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -g -fwrapv -O2 -Wl,-Bsymbolic-functions -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.10/common.o build/temp.linux-armv7l-3.10/memory.o build/temp.linux-armv7l-3.10/mq.o build/temp.linux-armv7l-3.10/semaphore.o build/temp.linux-armv7l-3.10/sysv_ipc_module.o -o build/lib.linux-armv7l-3.10/sysv_ipc.cpython-310-arm-linux-gnueabihf.so
      /usr/bin/ld: cannot find -lgcc: No such file or directory
      /usr/bin/ld: cannot find -lgcc: No such file or directory
      collect2: error: ld returned 1 exit status
      error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> sysv-ipc

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.`

### Description

Seemingly fails when trying to install `sysv-ipc`

### Additional information

Im using ubuntu image burned into SD card, ive already done a `apt update` + `apt-get upgrade`

TheRegularDX, 2024-08-14:
Oh also ive unminimized the system and tried again but still nothing

TheRegularDX, 2024-08-14:
now ive installed all of these possible missing packages via sudo
`python3-sysv-ipc gcc-arm-linux-gnueabihf build-essential python3-dev`
also created a venv
still gives the same error

TheRegularDX, 2024-08-14:
version `8.46.0` also fails to install but gives a different error related to ninja

TheRegularDX, 2024-08-14:
`sysv-ipc` simply fails to install every time, what even is this thing and what is its use in circuitpython...

makermelissa, 2024-08-14:
It's only used by a few boards for PulseIn and isn't really needed for your board. We probably should update the setup to only install for the boards that need it.

TheRegularDX, 2024-08-15:
thanks for the answer melissa!!!
ive downloaded the repo, removed `sysv-ipc` from the setup and installed it from there and it has been installed succesfully, although i have no idea if it works.
[image]

TheRegularDX, 2024-08-15:
it doesnt work, of course it doesnt

makermelissa, 2024-08-15:
> it from there and it has been installed succesfully, although i have no idea if it works.

What are you trying that doesn't work?

TheRegularDX, 2024-08-15:
> > it from there and it has been installed succesfully, although i have no idea if it works.
> 
> What are you trying that doesn't work?

running python and doing `import board`  returns this error:
```
ImportError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pico/Code/.devenv/lib/python3.10/site-packages/board.py", line 417, in <module>
    from adafruit_blinka.board.luckfox.luckfoxpico_mini import *
  File "/home/pico/Code/.devenv/lib/python3.10/site-packages/adafruit_blinka/board/luckfox/luckfoxpico_mini.py", line 6, in <module>
    from adafruit_blinka.microcontroller.rockchip.rv1103 import pin
  File "/home/pico/Code/.devenv/lib/python3.10/site-packages/adafruit_blinka/microcontroller/rockchip/rv1103/pin.py", line 5, in <module>
    from adafruit_blinka.microcontroller.generic_linux.libgpiod_pin import Pin
  File "/home/pico/Code/.devenv/lib/python3.10/site-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 8, in <module>
    raise ImportError(
ImportError: libgpiod Python bindings not found, please install and try again! See https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/libgpiod.py
```

makermelissa, 2024-08-15:
Try installing the libgpiod bindings with `pip3 install gpiod` or if that doesn't work, you can try installing python3-libgpiod.

TheRegularDX, 2024-08-15:
> Try installing the libgpiod bindings with `pip3 install gpiod` or if that doesn't work, you can try installing python3-libgpiod.

tried both, none worked, installing via pip returns this error:
```
Collecting gpiod
  Using cached gpiod-2.2.1.tar.gz (53 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: gpiod
  Building wheel for gpiod (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for gpiod (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [46 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-armv7l-cpython-310
      creating build/lib.linux-armv7l-cpython-310/gpiod
      copying gpiod/chip.py -> build/lib.linux-armv7l-cpython-310/gpiod
      copying gpiod/edge_event.py -> build/lib.linux-armv7l-cpython-310/gpiod
      copying gpiod/line_settings.py -> build/lib.linux-armv7l-cpython-310/gpiod
      copying gpiod/__init__.py -> build/lib.linux-armv7l-cpython-310/gpiod
      copying gpiod/version.py -> build/lib.linux-armv7l-cpython-310/gpiod
      copying gpiod/internal.py -> build/lib.linux-armv7l-cpython-310/gpiod
      copying gpiod/line.py -> build/lib.linux-armv7l-cpython-310/gpiod
      copying gpiod/exception.py -> build/lib.linux-armv7l-cpython-310/gpiod
      copying gpiod/info_event.py -> build/lib.linux-armv7l-cpython-310/gpiod
      copying gpiod/chip_info.py -> build/lib.linux-armv7l-cpython-310/gpiod
      copying gpiod/line_request.py -> build/lib.linux-armv7l-cpython-310/gpiod
      copying gpiod/line_info.py -> build/lib.linux-armv7l-cpython-310/gpiod
      running build_ext
      building 'gpiod._ext' extension
      creating build/temp.linux-armv7l-cpython-310
      creating build/temp.linux-armv7l-cpython-310/gpiod
      creating build/temp.linux-armv7l-cpython-310/gpiod/ext
      creating build/temp.linux-armv7l-cpython-310/lib
      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -D_GNU_SOURCE=1 -Iinclude -Ilib -Igpiod/ext -I/home/pico/Code/.devenv/include -I/usr/include/python3.10 -c gpiod/ext/chip.c -o build/temp.linux-armv7l-cpython-310/gpiod/ext/chip.o -Wall -Wextra -DGPIOD_VERSION_STR=\"2.1.3\"
      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -D_GNU_SOURCE=1 -Iinclude -Ilib -Igpiod/ext -I/home/pico/Code/.devenv/include -I/usr/include/python3.10 -c gpiod/ext/common.c -o build/temp.linux-armv7l-cpython-310/gpiod/ext/common.o -Wall -Wextra -DGPIOD_VERSION_STR=\"2.1.3\"
      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -D_GNU_SOURCE=1 -Iinclude -Ilib -Igpiod/ext -I/home/pico/Code/.devenv/include -I/usr/include/python3.10 -c gpiod/ext/line-config.c -o build/temp.linux-armv7l-cpython-310/gpiod/ext/line-config.o -Wall -Wextra -DGPIOD_VERSION_STR=\"2.1.3\"
      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -D_GNU_SOURCE=1 -Iinclude -Ilib -Igpiod/ext -I/home/pico/Code/.devenv/include -I/usr/include/python3.10 -c gpiod/ext/line-settings.c -o build/temp.linux-armv7l-cpython-310/gpiod/ext/line-settings.o -Wall -Wextra -DGPIOD_VERSION_STR=\"2.1.3\"
      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -D_GNU_SOURCE=1 -Iinclude -Ilib -Igpiod/ext -I/home/pico/Code/.devenv/include -I/usr/include/python3.10 -c gpiod/ext/module.c -o build/temp.linux-armv7l-cpython-310/gpiod/ext/module.o -Wall -Wextra -DGPIOD_VERSION_STR=\"2.1.3\"
      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -D_GNU_SOURCE=1 -Iinclude -Ilib -Igpiod/ext -I/home/pico/Code/.devenv/include -I/usr/include/python3.10 -c gpiod/ext/request.c -o build/temp.linux-armv7l-cpython-310/gpiod/ext/request.o -Wall -Wextra -DGPIOD_VERSION_STR=\"2.1.3\"      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -D_GNU_SOURCE=1 -Iinclude -Ilib -Igpiod/ext -I/home/pico/Code/.devenv/include -I/usr/include/python3.10 -c lib/chip-info.c -o build/temp.linux-armv7l-cpython-310/lib/chip-info.o -Wall -Wextra -DGPIOD_VERSION_STR=\"2.1.3\"
      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -D_GNU_SOURCE=1 -Iinclude -Ilib -Igpiod/ext -I/home/pico/Code/.devenv/include -I/usr/include/python3.10 -c lib/chip.c -o build/temp.linux-armv7l-cpython-310/lib/chip.o -Wall -Wextra -DGPIOD_VERSION_STR=\"2.1.3\"
      arm-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -D_GNU_SOURCE=1 -Iinclude -Ilib -Igpiod/e [... cut ...]
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for gpiod
Failed to build gpiod
ERROR: Could not build wheels for gpiod, which is required to install pyproject.toml-based projects

```

installing via `sudo apt install python3-libgpiod` returns:
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-libgpiod is already the newest version (1.6.3-1build1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

```

makermelissa, 2024-08-15:
> python3-libgpiod is already the newest version (1.6.3-1build1).

Ok, sounds like it's already installed. Kind of odd that it's giving you an error. What happens if you go into python and type `import gpiod` at the REPL?

TheRegularDX, 2024-08-15:
> > python3-libgpiod is already the newest version (1.6.3-1build1).
> 
> Ok, sounds like it's already installed. Kind of odd that it's giving you an error. What happens if you go into python and type `import gpiod` at the REPL?

it says that gpiod is not found, but when i run python as sudo it doesnt raise any errors

TheRegularDX, 2024-08-15:
ive uninstalled blinka from the venv, installed it again system wide via sudo, and importing board gives me a permission error 13. So i ran python as sudo and imported board, and it gives no errors

TheRegularDX, 2024-08-15:
heh, it seems `G34`, the pin of the board's built in LED is unavailable, im not at home right now (and wont be for the next 3 days) so i dont have any leds to test the classic blinky example. but it does seem to work, if i try it it with pin `G4` it gives no errors and i can run blinky with no problems, of course i cant see if its working 
because i dont have any leds so ill just asume it does work for now.

TheRegularDX, 2024-08-16:
i should have said this yesterday before going to sleep but i feel like sudo shouldnt be necessary for this, just sayin'

TheRegularDX, 2024-08-20:
Got it working
https://github.com/user-attachments/assets/8b1d2aad-27ef-45ef-a991-47e84703bb52

closed 2024-08-20 by TheRegularDX

nopnop2002, 2025-04-16:
@makermelissa 

I am having the same problem.
Failed to build sysv_ipc.
```
$ python3 -m pip install Adafruit-Blinka

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for sysv_ipc
  Running setup.py clean for sysv_ipc
Failed to build sysv_ipc
ERROR: Failed to build installable wheels for some pyproject.toml based projects (sysv_ipc)
```

>It's only used by a few boards for PulseIn and isn't really needed for your board. We probably should update the setup to only install for the boards that need it.

Is there an option to not use sysv-ipc?

```
$ python3 -m pip install Adafruit-Blinka --help

Usage:
  /usr/bin/python3 -m pip install [options] <requirement specifier> [package-index-options] ...
  /usr/bin/python3 -m pip install [options] -r <requirements file> [package-index-options] ...
  /usr/bin/python3 -m pip install [options] [-e] <vcs project url> ...
  /usr/bin/python3 -m pip install [options] [-e] <local project path> ...
  /usr/bin/python3 -m pip install [options] <archive url/path> ...

Description:
  Install packages from:

  - PyPI (and other indexes) using requirement specifiers.
  - VCS project urls.
  - Local project directories.
  - Local or remote source archives.

  pip also supports installing from "requirements files", which provide
  an easy way to specify a whole environment to be installed.

Install Options:
  -r, --requirement <file>    Install from the given requirements file. This
                              option can be used multiple times.
  -c, --constraint <file>     Constrain versions using the given constraints
                              file. This option can be used multiple times.
  --no-deps                   Don't install package dependencies.
  --pre                       Include pre-release and development versions. By
                              default, pip only finds stable versions.
  -e, --editable <path/url>   Install a project in editable mode (i.e.
                              setuptools "develop mode") from a local project
                              path or a VCS url.
  --dry-run                   Don't actually install anything, just print what
                              would be. Can be used in combination with
                              --ignore-installed to 'resolve' the
                              requirements.
  -t, --target <dir>          Install packages into <dir>. By default this
                              will not replace existing files/folders in
                              <dir>. Use --upgrade to replace existing
                              packages in <dir> with new versions.
  --platform <platform>       Only use wheels compatible with <platform>.
                              Defaults to the platform of the running system.
                              Use this option multiple times to specify
                              multiple platforms supported by the target
                              interpreter.
  --python-version <python_version>
                              The Python interpreter version to use for wheel
                              and "Requires-Python" compatibility checks.
                              Defaults to a version derived from the running
                              interpreter. The version can be specified using
                              up to three dot-separated integers (e.g. "3" for
                              3.0.0, "3.7" for 3.7.0, or "3.7.3"). A major-
[... 164 lines cut ...]
                              (s)witch, (i)gnore, (w)ipe, (b)ackup, (a)bort.
  --trusted-host <hostname>   Mark this host or host:port pair as trusted,
                              even though it does not have valid or any HTTPS.
  --cert <path>               Path to PEM-encoded CA certificate bundle. If
                              provided, overrides the default. See 'SSL
                              Certificate Verification' in pip documentation
                              for more information.
  --client-cert <path>        Path to SSL client certificate, a single file
                              containing the private key and the certificate
                              in PEM format.
  --cache-dir <dir>           Store the cache data in <dir>.
  --no-cache-dir              Disable the cache.
  --disable-pip-version-check
                              Don't periodically check PyPI to determine
                              whether a new version of pip is available for
                              download. Implied with --no-index.
  --no-color                  Suppress colored output.
  --no-python-version-warning
                              Silence deprecation warnings for upcoming
                              unsupported Pythons.
  --use-feature <feature>     Enable new functionality, that may be backward
                              incompatible.
  --use-deprecated <feature>  Enable deprecated functionality, that will be
                              removed in the future.
```