adafruit/Adafruit_Blinka #886: Beagle Bone Black: no such file or directory: 0-00500/nvmem, permission denied 0-00500/nvmem

trparry, 2024-08-16:
### Board Name

Beagle Bone Black Rev C

### Steps

Hello, I am attempting to install Adafruit Blinka on a beagle bone black.

Here is my info:
Image: Debian 12.2 - Bookworm IoT Image 2023-10-07
eeprom:[A335BNLT00C02345SBB00479] (got this using sudo beagle-version)
Python 3.11
Adafruit-Blinka version 3.1.1
Adafruit Platform-Detect version 3.73.0
Adafruit python shell 1.8.1

My error is very similar to #687 and also I have one of the errors shown in #283. 

When I try to import board I get two errors:

Traceback (most recent call last):
  File "/home/debian/.venvs/MyEnv/lib/python3.11/site-packages/adafruit_platformdetect/board.py", line 320, in _beaglebone_id
    with open("/sys/bus/nvmem/devices/0-00500/nvmem", "rb") as eeprom:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/sys/bus/nvmem/devices/0-00500/nvmem'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/debian/blinkatest.py", line 2, in <module>
    import board
  File "/home/debian/.venvs/MyEnv/lib/python3.11/site-packages/board.py", line 21, in <module>
    from adafruit_blinka.agnostic import board_id, detector
  File "/home/debian/.venvs/MyEnv/lib/python3.11/site-packages/adafruit_blinka/agnostic/__init__.py", line 22, in <module>
    board_id = detector.board.id
               ^^^^^^^^^^^^^^^^^
  File "/home/debian/.venvs/MyEnv/lib/python3.11/site-packages/adafruit_platformdetect/board.py", line 74, in id
    board_id = self._beaglebone_id()
               ^^^^^^^^^^^^^^^^^^^^^
  File "/home/debian/.venvs/MyEnv/lib/python3.11/site-packages/adafruit_platformdetect/board.py", line 324, in _beaglebone_id
    with open("/sys/bus/nvmem/devices/0-00501/nvmem", "rb") as eeprom:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/sys/bus/nvmem/devices/0-00501/nvmem'

I also tried to add my board ID to the list as suggested in #283 at site-packages/adafruit_platformdetect/constants but this does not seem to change the error. Perhaps I need to recompile the package or something?

trparry, 2024-08-19:
I found the problem: 

Because I can't say '''sudo pip3 install Adafruit_BBIO'' because of the externally managed error, I was forced to say: '''pip3 install Adafruit_BBIO'''. 

This then caused the warning:
 ''' DEPRECATION: Adafruit_BBIO is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559 '''

and the enventual error:

''' error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> Adafruit_BBIO '''

So, I had to install Adafruit_BBIO system wide (with sudo) manually as described here: https://github.com/adafruit/adafruit-beaglebone-io-python

Then I said sudo python3 blinka.py (my test script) and everything works properly. 

I think originally my first error was because Adafruit_BBIO was not installed system wide, and my second error was because I wasn't running blinka.py as root giving the second permission denied error. 

The whole "externally managed environment error" is really messing with me. It's described here: https://askubuntu.com/questions/1465218/pip-error-on-ubuntu-externally-managed-environment-%C3%97-this-environment-is-extern

I'm going to close this.

closed 2024-08-19 by trparry

makermelissa, 2024-08-19:
> The whole "externally managed environment error" is really messing with me.

Me too. :) This guide does help: https://learn.adafruit.com/python-virtual-environment-usage-on-raspberry-pi even though it says Raspberry Pi.

bgigous, 2024-11-26:
(Hi @makermelissa ! Thank for your help. Nice to see your comment. We met at PyCon 2021! :)

Honestly, I don't think being forced to install Adafruit_Blinka systemwide just to run `python3 -c "import board"` is really a solution (on top of that, it requires `sudo` which is quite a workaround). I'd argue this issue be reopened.

---

For others installing this package on the BeagleBone **Green** platform, I was met with more resistance than @trparry with the same steps, which were:

1. Install Adafruit_BBIO systemwide with `sudo python3 setup.py install`
2. Install Adafruit_Blinka systemwide with `sudo python3 setup.py install`
3. Run `sudo python3 blinka.py` where `blinka.py` simply contains `import board`

Ideally, running the script in step 3 would run without error, but I actually got stuck at step 2. With `python3-setuptools` debian package fully upgraded to v66.1.1-1 with `sudo apt install --upgrade python3-setuptools` on the same OS as @trparry , I got this error after running `setup.py install` for Adafruit_Blinka:

```
  # ...
  File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 107, in __getattr__
    raise AttributeError(attr)
AttributeError: install_layout. Did you mean: 'install_platlib'?
```

To fix it, I actually temporarily downgraded setuptools to v60.5.0 and reran setup.py for Adafruit_Blinka. However, I believe setting this environment variable before running setup.py works as well to get around this error:

```bash
export SETUPTOOLS_USE_DISTUTILS=stdlib
```

bgigous, 2024-11-26:
After adding a udev rule for eeprom/nvmem, I'm now able to run `python -c "import board"` without sudo.

I found this issue: https://github.com/adafruit/Adafruit_Python_PlatformDetect/issues/218

The issue seems to be for BeagleBone Black on Debian Buster, but the comments gave me enough of a hint.

I had to make a new file `/etc/udev/rules.d/80-eeprom-noroot.rules` with these contents:
```
# See https://github.com/adafruit/Adafruit_Python_PlatformDetect/issues/218
# COPIED FROM https://github.com/s-light/pocketbeagle_python_tests/blob/master/80-eeprom-noroot.rules
# MODIFIED: DEVPATH=="*0-00500" -> DEVPATH=="*0-00500"
# ORIGINAL COMMENT:
# /etc/udev/rules.d/80-eeprom-noroot.rules
#
# allow i2c group to read of eeprom
# based on:
# https://martin-jones.com/2016/08/12/making-beaglebone-black-serial-number-available-to-user-applications/
# extended by: Stefan Krüger
#
# allows both locations to be read by the i2c group:
# /sys/bus/i2c/devices/0-0050/eeprom
# /sys/bus/nvmem/devices/0-00500/nvmem
#
# check result:
# ls -l /sys/bus/i2c/devices/0-0050/eeprom /sys/bus/nvmem/devices/0-00500/nvmem
# both should be readable. similar to this
# -rw-r----- 1 root i2c 0 XXXXXXXXX
#

SUBSYSTEM=="i2c", DEVPATH=="*0-0050", RUN+="/bin/chown root:i2c /sys$env{DEVPATH}/eeprom", RUN+="/bin/chmod 0640 /sys$env{DEVPATH}/eeprom"
SUBSYSTEM=="nvmem", DEVPATH=="*0-00501", RUN+="/bin/chown root:i2c /sys$env{DEVPATH}/nvmem", RUN+="/bin/chmod 0640 /sys$env{DEVPATH}/nvmem"
```