adafruit/Adafruit_Blinka #903: FT232H not working in Thonny but can be seen in terminal

sjagrant, 2024-10-16:
FT232H not working in Thonny but can be seen in terminal.  Running on Windows and followed Adafruit install instructions with positive feedback for tests.  

 raise NotImplementedError(
NotImplementedError: 
        Adafruit-PlatformDetect version 3.75.0 was unable to identify the board and/or
        microcontroller running the Windows platform. Please be sure you
        have the latest packages by running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect' 

Check and installed correctly

caternuson, 2024-10-16:
Probably related to Thonny not seeing the environment variable.

Did you run the post install checks from within Thonny also?
https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/troubleshooting
In particular, this one:
https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/troubleshooting#check-environment-variable-within-python-3041702

caternuson, 2024-10-16:
* In Thonny, go to Tools -> Options...
* Enter `BLINKA_FT232H=1` in the Environment variables
* Restart Thonny

[image]

Can test like this:
[image]

sjagrant, 2024-10-16:
Many thanks Carter that sorted it out.

What excellent support.
Best regards,
stuart

sjagrant, 2024-10-16:
Yes I did,  but when I got to  NotImplementedError recheck Zadig I didn't
get any further and kept going back to setup steps.---

-------------------------------------

If you get

NotImplementedError: Operation not supported or unimplemented on this
platform

Go back to the setup steps and Zadig your board if on Windows
<https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/troubleshooting#check-environment-variable-within-python-3041702>
Check environment variable within python

At the Python REPL, type

import os
os.environ["BLINKA_FT232H"]

if you get a KeyError it means you did not set the environment variable
right

------------------------------------------------

Again many thanks for your great support.

caternuson, 2024-11-07:
Closing since resolved.

closed 2024-11-07 by caternuson