adafruit/Adafruit_Blinka #907: Cannot determine SOC peripheral base address

caternuson, 2024-11-07:
### Board Name

Pi 5

### Steps

Follow the guide for initial setup:
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi
and then run the blinkatest.py script.

### Description

The installation runs without issue. But when the blinkatest.py script is run, get the following error:
```
(env) pi@raspberrypi:~ $ python3 blinkatest.py
Hello, blinka!
Traceback (most recent call last):
  File "/home/pi/blinkatest.py", line 8, in <module>
    pin = digitalio.DigitalInOut(board.D4)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/env/lib/python3.11/site-packages/digitalio.py", line 197, in __init__
    self.direction = Direction.INPUT
    ^^^^^^^^^^^^^^
  File "/home/pi/env/lib/python3.11/site-packages/digitalio.py", line 227, in direction
    self._pin.init(mode=Pin.IN)
  File "/home/pi/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pin.py", line 40, in init
    GPIO.setup(self.id, GPIO.IN)
RuntimeError: Cannot determine SOC peripheral base address
(env) pi@raspberrypi:~ $
```

Bichidian, 2024-11-08:
I had the same problem. I figured out that this was caused by using the RPi.GPIO package, which does not support Raspberry Pi 5. The strange thing is that I found this bug was already fixed by pull request #856. Reinstalling directly from github by `pip install git+https://github.com/adafruit/Adafruit_Blinka.git` fixed the problem for me. Somehow `pip install Adafruit_Blinka` got the wrong dependency.

caternuson, 2024-11-08:
Since it's pulling the pin from `bcm283x`, it seems like the detector is not correctly IDing the board as a Pi5.
https://github.com/adafruit/Adafruit_Blinka/blob/b91d3fc70bd1599ea50517a390d18a3fc455fc85/src/digitalio.py#L17-L27

caternuson, 2024-11-08:
Or something else?
```
(env) pi@raspberrypi:~ $ python3
Python 3.11.2 (main, Aug 26 2024, 07:20:54) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import digitalio
>>> digitalio.detector.chip.BCM2XXX
True
>>> digitalio.board_id
'RASPBERRY_PI_5'
>>> digitalio.Pin
<class 'adafruit_blinka.microcontroller.bcm283x.pin.Pin'>
>>> 
```

caternuson, 2024-11-08:
Oh, that's just what it does:
https://github.com/adafruit/Adafruit_Blinka/blob/b91d3fc70bd1599ea50517a390d18a3fc455fc85/src/adafruit_blinka/microcontroller/bcm2711/pin.py#L4-L6

MGPhil, 2024-11-08:
> I had the same problem. I figured out that this was caused by using the RPi.GPIO package, which does not support Raspberry Pi 5. The strange thing is that I found this bug was already fixed by pull [...]

> I had the same problem. I figured out that this was caused by using the RPi.GPIO package, which does not support Raspberry Pi 5. The strange thing is that I found this bug was already fixed by pull [...]

For me RPi.GPIO works out of the box on PI 5, install Blinka or rpi-lgpio breaks the GPIO function

caternuson, 2024-11-08:
There's a hardwired pip install of RPi.GPIO in the installer script:
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/ad19b39863503ce5edf76cfff1334ccad388c03d/raspi-blinka.py#L95
which may be interfering with the rpi-lgpio install?

https://pypi.org/project/rpi-lgpio/

> You cannot install rpi-lgpio and [rpi-gpio](https://pypi.org/project/RPi.GPIO/) (aka RPi.GPIO, the library it emulates) at the same time, in the same Python environment. Both packages attempt to install a module named RPi.GPIO and obviously this will not work.

MGPhil, 2024-11-08:
> There's a hardwired pip install of RPi.GPIO in the installer script: https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/ad19b39863503ce5edf76cfff1334ccad388c03d/raspi-blinka.py#L95 [...]

So how to install blinka without rpi-lgpio as RPI.GPIO works out of the box now on PI 5?

caternuson, 2024-11-08:
Tested above by manually uninstalling RPi.GPIO:
```
(env) pi@raspberrypi:~ $ pip3 uninstall RPi.GPIO
Found existing installation: RPi.GPIO 0.7.1
Uninstalling RPi.GPIO-0.7.1:
  Would remove:
    /home/pi/env/lib/python3.11/site-packages/RPi.GPIO-0.7.1.dist-info/*
    /home/pi/env/lib/python3.11/site-packages/RPi/*
Proceed (Y/n)? y
  Successfully uninstalled RPi.GPIO-0.7.1
```
And then it works:
```
(env) pi@raspberrypi:~ $ python3
Python 3.11.2 (main, Aug 26 2024, 07:20:54) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> import digitalio
>>> pin = digitalio.DigitalInOut(board.D4)
>>> 
```

caternuson, 2024-11-08:
> So how to install blinka without rpi-lgpio as RPI.GPIO works out of the box now on PI 5?

@MGPhil You don't. There's an installation issue (and maybe more) that needs to be fixed. Currently investigating and documenting here. This is not a user issue. It needs to be fixed by Adafruit.

MGPhil, 2024-11-08:
@caternuson OK then I wait for the fix, just saying I postet that issue in the adafruit forum and debugged a lot before that post. As I found out RPi.GPIO works out of the box on PI 5 for me and thats why rpi.lgpio which is installed with blinka breaks both.

caternuson, 2025-01-21:
@mikeysklar What happens if you pip install a CircuitPython library on that same setup and then try running `blinkatest.py` again? I think this issue is more related to a library conflict between `rpi-lgpio` and `RPi.GPIO`. Only one should end up installed: `rpi-lgpio` for Pi5 and `RPi.GPIO` for all others. The initial installer script gets things setup correctly on Pi5. However, pip installing then brings back `RPi.GPIO` for some reason, which causes this error to show up again.

mikeysklar, 2025-01-21:
right you are:

```
pi@pi5:~ $ source env/bin/activate
(env) pi@pi5:~ $ python3 ./blinkatest.py
Hello, blinka!
Digital IO ok!
I2C ok!
SPI ok!
done!
(env) pi@pi5:~ $ pip3 install adafruit-circuitpython-epd
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting adafruit-circuitpython-epd
  Downloading https://www.piwheels.org/simple/adafruit-circuitpython-epd/adafruit_circuitpython_epd-2.13.2-py3-none-any.whl (31 kB)
Requirement already satisfied: Adafruit-Blinka in ./env/lib/python3.11/site-packages (from adafruit-circuitpython-epd) (8.50.0)
Requirement already satisfied: adafruit-circuitpython-busdevice in ./env/lib/python3.11/site-packages (from adafruit-circuitpython-epd) (5.2.10)
Collecting adafruit-circuitpython-framebuf
  Downloading https://www.piwheels.org/simple/adafruit-circuitpython-framebuf/adafruit_circuitpython_framebuf-1.6.7-py3-none-any.whl (9.2 kB)
Requirement already satisfied: typing-extensions~=4.0 in /usr/lib/python3/dist-packages (from adafruit-circuitpython-epd) (4.4.0)
Requirement already satisfied: Adafruit-PlatformDetect>=3.70.1 in ./env/lib/python3.11/site-packages (from Adafruit-Blinka->adafruit-circuitpython-epd) (3.76.1)
Requirement already satisfied: Adafruit-PureIO>=1.1.7 in ./env/lib/python3.11/site-packages (from Adafruit-Blinka->adafruit-circuitpython-epd) (1.1.11)
Requirement already satisfied: binho-host-adapter>=0.1.6 in ./env/lib/python3.11/site-packages (from Adafruit-Blinka->adafruit-circuitpython-epd) (0.1.6)
Requirement already satisfied: pyftdi>=0.40.0 in ./env/lib/python3.11/site-packages (from Adafruit-Blinka->adafruit-circuitpython-epd) (0.56.0)
Requirement already satisfied: adafruit-circuitpython-typing in ./env/lib/python3.11/site-packages (from Adafruit-Blinka->adafruit-circuitpython-epd) (1.11.2)
Collecting RPi.GPIO
  Using cached RPi.GPIO-0.7.1-cp311-cp311-linux_aarch64.whl
Requirement already satisfied: rpi-ws281x>=4.0.0 in ./env/lib/python3.11/site-packages (from Adafruit-Blinka->adafruit-circuitpython-epd) (5.0.0)
Requirement already satisfied: sysv-ipc>=1.1.0 in ./env/lib/python3.11/site-packages (from Adafruit-Blinka->adafruit-circuitpython-epd) (1.1.0)
Requirement already satisfied: pyserial in /usr/lib/python3/dist-packages (from binho-host-adapter>=0.1.6->Adafruit-Blinka->adafruit-circuitpython-epd) (3.5)
Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in ./env/lib/python3.11/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-epd) (1.3.1)
Requirement already satisfied: adafruit-circuitpython-requests in ./env/lib/python3.11/site-packages (from adafruit-circuitpython-typing->Adafruit-Blinka->adafruit-circuitpython-epd) (4.1.8)
Requirement already satisfied: Adafruit-Circuitpython-ConnectionManager in ./env/lib/python3.11/site-packages (from adafruit-circuitpython-requests->adafruit-circuitpython-typing->Adafruit-Blinka->adafruit-circuitpython-epd) (3.1.2)
Installing collected packages: RPi.GPIO, adafruit-circuitpython-framebuf, adafruit-circuitpython-epd
Successfully installed RPi.GPIO-0.7.1 adafruit-circuitpython-epd-2.13.2 adafruit-circuitpython-framebuf-1.6.7
(env) pi@pi5:~ $ python3 ./blinkatest.py
Hello, blinka!
Traceback (most recent call last):
  File "/home/pi/./blinkatest.py", line 8, in <module>
    pin = digitalio.DigitalInOut(board.D4)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/env/lib/python3.11/site-packages/digitalio.py", line 197, in __init__
    self.direction = Direction.INPUT
    ^^^^^^^^^^^^^^
  File "/home/pi/env/lib/python3.11/site-packages/digitalio.py", line 227, in direction
    self._pin.init(mode=Pin.IN)
  File "/home/pi/env/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pin.py", line 40, in init
    GPIO.setup(self.id, GPIO.IN)
RuntimeError: Cannot determine SOC peripheral base address
```

caternuson, 2025-01-21:
Thanks for checking/confirming.

If you manually pip uninstall RPi.GPIO it should get back to working again.
```
pip3 uninstall -y RPi.GPIO
```

This behavior is known, but hasn't been resolved yet.

mikeysklar, 2025-01-21:
Right again...

```
(env) pi@pi5:~ $ pip3 uninstall -y RPi.GPIO
Found existing installation: RPi.GPIO 0.7.1
Uninstalling RPi.GPIO-0.7.1:
  Successfully uninstalled RPi.GPIO-0.7.1
(env) pi@pi5:~ $ python3 ./blinkatest.py
Hello, blinka!
Digital IO ok!
I2C ok!
SPI ok!
done!
```

mikeysklar, 2025-01-28:
I've not tried this variation yet, but the Blinka installer is using this from a [PR #324 I did in November](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/pull/324) to get blinkatest.py to pass.

```
sudo apt remove python3-rpi.gpio
pip3 install rpi-lgpio
```

Maybe this would have been a better choice to get keep apt with apt. 

```
sudo apt autoremove python3-rpi.gpio
sudo apt install python3-rpi-lgpio
```

This would not resolve RPi.GPIO getting re-installed each time with pip3 since the CircuitPython packages list it as as dependency.

makermelissa, 2025-01-29:
The issue appears to finally be resolved by https://github.com/adafruit/Adafruit_Blinka/pull/936 and as of Blinka 8.52.0, it should install correctly.

closed 2025-01-29 by makermelissa