adafruit/Adafruit_Blinka #916: `uv pip` fails to install necessary packages

jepler, 2024-12-10:
### Board Name

Raspberry Pi 5

### Steps

 1. Install `uv`
 2. `uv venv /tmp/blinka-uv`
 3. `. /tmp/blinka-uv/bin/activate`
 4. `uv pip install adafruit-blinka`
 5. `python -c 'import board`

### Description

Following these steps, the result is `ModuleNotFoundError: No module named 'RPi'`. Comparing `uv pip freeze` in this venv to `pip freeze` in a working venv, one can see the following two files were not installed: `rpi-ws281x==5.0.0 RPi.GPIO==0.7.1`

### Additional information

This is likely to be related to how these declarations are conditional in the setup.py file. I couldn't find where this was a documented limitation of uv, but it seems possible/likely.

. o O ( I'm actually not sure how good old pip knows how to install RPi.GPIO when it installs a wheel of adafruit-blinka... )

jepler, 2024-12-10:
just documenting that blinka & uv are incompatible is probably fine for the moment!

caternuson, 2024-12-10:
This is semi-related to #907 and https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/pull/329

I think the only way to get it working (currently) is via that installer script, which does some manual installing/removing to get things setup for Pi5. (see PR 329)

So not surprising things generally aren't working with other tooling.

2bndy5, 2025-11-27:
This feels more like a problem with `RPi.GPIO` package. Its rather uncommon to have a dot in the package name. Before `uv` came around, I remember seeing this `No module named 'RPi'` error before; but it usually meant that the package `RPi.GPIO` wasn't installed (via `apt-get`).

BTW, the `RPi.GPIO` implementation does not work on RPi5 because the GPIO pins are abstracted to a separate RP1 chip. See [ticket on sourceforge](https://sourceforge.net/p/raspberry-gpio-python/tickets/214/) (which I think makermelissa raised).

makermelissa, 2026-10-26:
Fixed by #1131, #1122, and #1153.

closed 2026-10-26 by makermelissa