adafruit/Adafruit_Blinka #919: Is the new Raspberry Pi CM5 Rev 1.0 board supported?

Sparqx73, 2024-12-20:
I am using the new Raspberry Pi 5 Compute IO Board with a CM5 and I am getting the following error when running a python script to display Raspberry Pi Stats on an Oled which uses the Adafruit Blinka Libraries:

python3 monitor.py
Traceback (most recent call last):
File "/home/pi/OLED_Stats/monitor.py", line 7, in
import board
File "/home/pi/stats_env/lib/python3.11/site-packages/board.py", line 451, in
raise NotImplementedError(
NotImplementedError:
Adafruit-PlatformDetect version 3.75.0 was unable to identify the board and/or
microcontroller running the Linux platform. Please be sure you
have the latest packages by running:
'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

    If you are running the latest package, your board may not yet be supported. Please
    open a New Issue on GitHub at https://github.com/adafruit/Adafruit_Blinka/issues and
    select New Board Request.
Are you able to advise whether the new Compute Module 5 is supported? Or when it will be supported?

Thanks in advance for your help

danic85, 2024-12-30:
@Sparqx73 I'm getting this on the CM5 as well. It looks like we don't have it in the list of supported boards yet.

https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/main/adafruit_platformdetect/constants/boards.py#L146

There is a guide to allow us to add them ourselves. I'm wondering if anyone has worked on this yet: https://learn.adafruit.com/adding-a-single-board-computer-to-platformdetect-for-blinka

Sparqx73, 2024-12-31:
> @Sparqx73 I'm getting this on the CM5 as well. It looks like we don't have it in the list of supported boards yet. [...]

I've worked on this with my brother and together we got it to work by amending the boards.py as follows:

# Various Raspberry Pi models
RASPBERRY_PI_B_REV1 = "RASPBERRY_PI_B_REV1"
RASPBERRY_PI_B_REV2 = "RASPBERRY_PI_B_REV2"
RASPBERRY_PI_B_PLUS = "RASPBERRY_PI_B_PLUS"
RASPBERRY_PI_A = "RASPBERRY_PI_A"
RASPBERRY_PI_A_PLUS = "RASPBERRY_PI_A_PLUS"
RASPBERRY_PI_CM1 = "RASPBERRY_PI_CM1"
RASPBERRY_PI_ZERO = "RASPBERRY_PI_ZERO"
RASPBERRY_PI_ZERO_W = "RASPBERRY_PI_ZERO_W"
RASPBERRY_PI_ZERO_2_W = "RASPBERRY_PI_ZERO_2_W"
RASPBERRY_PI_2B = "RASPBERRY_PI_2B"
RASPBERRY_PI_3B = "RASPBERRY_PI_3B"
RASPBERRY_PI_3B_PLUS = "RASPBERRY_PI_3B_PLUS"
RASPBERRY_PI_CM3 = "RASPBERRY_PI_CM3"
RASPBERRY_PI_3A_PLUS = "RASPBERRY_PI_3A_PLUS"
RASPBERRY_PI_CM3_PLUS = "RASPBERRY_PI_CM3_PLUS"
RASPBERRY_PI_4B = "RASPBERRY_PI_4B"
RASPBERRY_PI_AVNET_IIOT_GW = "RASPBERY_PI_AVNET_IIOT_GW"
RASPBERRY_PI_400 = "RASPBERRY_PI_400"
RASPBERRY_PI_CM4 = "RASPBERRY_PI_CM4"
RASPBERRY_PI_CM4S = "RASPBERRY_PI_CM4S"
RASPBERRY_PI_5 = "RASPBERRY_PI_5"
RASPBERRY_PI_CM5 = "RASPBERRY_PI_CM5"

_RASPBERRY_PI_CM_IDS = (
    RASPBERRY_PI_CM1,
    RASPBERRY_PI_CM3,
    RASPBERRY_PI_CM3_PLUS,
    RASPBERRY_PI_CM4,
    RASPBERRY_PI_CM4S,
    RASPBERRY_PI_CM5,

_PI_MODELS = {
    0x00: RASPBERRY_PI_A,
    0x01: {
        1.0: RASPBERRY_PI_B_REV1,
        2.0: RASPBERRY_PI_B_REV2,
    },
    0x02: RASPBERRY_PI_A_PLUS,
    0x03: RASPBERRY_PI_B_PLUS,
    0x04: RASPBERRY_PI_2B,
    0x06: RASPBERRY_PI_CM1,
    0x08: RASPBERRY_PI_3B,
    0x09: RASPBERRY_PI_ZERO,
    0x0A: RASPBERRY_PI_CM3,
    0x0B: RASPBERRY_PI_AVNET_IIOT_GW,
    0x0C: RASPBERRY_PI_ZERO_W,
    0x0D: RASPBERRY_PI_3B_PLUS,
    0x0E: RASPBERRY_PI_3A_PLUS,
    0x10: RASPBERRY_PI_CM3_PLUS,
    0x11: RASPBERRY_PI_4B,
    0x12: RASPBERRY_PI_ZERO_2_W,
    0x13: RASPBERRY_PI_400,
    0x14: RASPBERRY_PI_CM4,
    0x15: RASPBERRY_PI_CM4S,
    0x17: RASPBERRY_PI_5,
    0x18: RASPBERRY_PI_CM5,
}

I am in the midst of debriefing on what else we might have modified but this will be a good start if you haven't done it already...I will come back and add anything else that we adjusted once I have documented it

danic85, 2024-12-31:
Thanks for this @Sparqx73  - that worked for me too. I am using a virtual environment so my locations were different, but these changes resolved the error. :clap:

Sparqx73, 2024-12-31:
It looks like we were successful in adding the additional entries for the CM5 to make the Adafruit_Blinka libraries work for this new board so I will close this issue.

dhalbert, 2024-12-31:
If this works for you could you please submit one or more pull requests?

danic85, 2025-01-02:
@dhalbert PR created here: https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/374

PR #929 (2025-01-16): Add Pi500 and CM5 Support
This adds support for the Pi500, CM5, and CM5 lite. Tested on Pi500 including running Neopixels. Simplified some of the import board checks so future boards can just be added to platformdetect.

Requires https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/377.
Fixes #922. Fixes #919.