adafruit/Adafruit_Blinka #940: Add pio-based IncrementalEncoder on pi5

jepler, 2025-02-07:
In [rp1pio there is an example](https://github.com/adafruit/Adafruit_Blinka_Raspberry_Pi5_rp1pio/blob/main/examples/quadrature_encoder.py) of using the pio peripheral to implement quadrature counting that doesn't require any CPU time.

We would like to see this code adapted into Adafruit_Blinka, and used instead of the thread-based implementation added in #937 when appropriate.

Benefits/limitations:

- uses the pio peripheral, which means no CPU time is required between polls
- very high polling rates above 1MHz can be used.
- Can count up to +-2 billion counts (full 32 bit signed counter)
- up to 4 encoders can be connected in this way (3 if neopixel is used)
- encoder pins for one knob must be adjacent
- only works on pi5 and only when rp1pio is installed and properly configured

jepler, 2025-02-07:
ping @ladyada @makermelissa

ladyada, 2025-02-07:
neat 🎛️

makermelissa, 2025-02-11:
Awesome, I'll take a look and see if I can get that going.

makermelissa, 2025-03-14:
Fixed by #941.

closed 2025-03-14 by makermelissa