adafruit/Adafruit_Blinka #943: Freeze on RasPi 4 when importing board

dlayne-i2, 2025-02-18:
### Board Name

Raspberry Pi 4

### Steps

Hi,

I am encountering the exact same issue that another user encountered here:  https://github.com/adafruit/Adafruit_Blinka/issues/563

I have an adafruit MCP2221A board that I am controlling via a small Python script.  Everything works fine on windows.  However, when running on the Pi 4, the entire python runtime hangs when importing board.  The hang lasts indefinitely, but if I disconnect the MCP2221 while the issue is occurring, then I get an OSError: read error and the Python runtime exits

I have traced this hang down to a call to _hid_xfer on line 85, which is being called from gp_set_mode which is called from __init__ in mcp2221.py

Note some important details:
-This MCP2221 has been converted to 5V logic level from the base 3.3V.  I have a second board which is still at 3.3V, and this one works just fine on the Pi.  It is only the board has been converted to 5V that fails (though both boards work fine on a windows machine)
-I went through the setup guide and all sanity checks pass.  The board is identified properly if I list USB devices

Unfortunately for the project I am working on I require the 5V logic level (the DAC is being used to drive a power supply to dim some LEDs), so I need to get the 5V board working properly when connected to the RasPi 4.

Thanks

makermelissa, 2025-03-14:
Follow [this guide](https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221/linux) if you haven't already and make sure to call `export BLINKA_MCP2221=1` as that should tell it you want to run it on the MCP2221a. It might be getting confused about whether you are intending to run Blinka on the Pi 4 or the MCP2221. However, based on you unplugging it, it sounds like you might have that sorted out already.

caternuson, 2025-06-05:
Do all of the post install checks pass?
https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221/post-install-checks

makermelissa, 2025-08-27:
Closing due to lack of response.

closed 2025-08-27 by makermelissa