adafruit/Adafruit_Blinka #951: Jetso Orin Nano

sarahalipour, 2025-03-09:
Please add support for jetson orin nano
I've already installed the latest version on jetpack 6.2 using `pip3 install --upgrade adafruit-blinka adafruit-platformdetect`

Here is the reproducible exampe:
`from adafruit_servokit import ServoKit`
which gives below error:
`from adafruit_servokit import ServoKit
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.10/dist-packages/adafruit_servokit.py", line 35, in <module>
    import board
  File "/home/sahand/.local/lib/python3.10/site-packages/board.py", line 445, in <module>
    raise NotImplementedError(
NotImplementedError: 
        Adafruit-PlatformDetect version 3.77.0 was unable to identify the board and/or
        microcontroller running the Linux platform. `
python version == 3.10.12

chpayan, 2025-03-12:
I'm experiencing the same issue. I'm running a simple Python test program as follows:

```python
import board
import digitalio
import busio
print("Hello Blinka!")
```

Upon execution, I receive the following error:

```
Traceback (most recent call last):
  File "/home/postdoc/Desktop/blinkatest.py", line 1, in <module>
    import board
  File "/home/postdoc/.local/lib/python3.10/site-packages/board.py", line 445, in <module>
    raise NotImplementedError(
NotImplementedError: 
        Adafruit-PlatformDetect version 3.77.0 was unable to identify the board and/or
        microcontroller running the Linux platform. Please ensure you
        have the latest packages installed by running:

        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

        If you are already using the latest version, your board may not yet be supported.
        In that case, consider opening a New Issue on GitHub at:
        https://github.com/adafruit/Adafruit_Blinka/issues 
        and select "New Board Request."
```

Any insights or suggestions would be greatly appreciated! If this is not a platform support issue, what other potential causes should I investigate?

SkrilaxCZ, 2025-03-17:
Same problem. I'm guessing this has to do with Super mode on Orin Nano / NX?

SkrilaxCZ, 2025-03-17:
Yep, in `($CONDA_PREFIX)/lib/python3.xx/site-packages/adafruit_platformdetect/constants/board.py`, find ORIN_NANO, then pick the board that you have (devkit: "nvidia,p3768-0000+p3767-0005") and add a new line with "nvidia,p3768-0000+p3767-0005-super". Solved

chpayan, 2025-03-17:
> Yep, in `($CONDA_PREFIX)/lib/python3.xx/site-packages/adafruit_platformdetect/constants/board.py`, find ORIN_NANO, then pick the board that you have (devkit: `nvidia,p3768-0000+p3767-0005`) and add a new line with `nvidia,p3768-0000+p3767-0005-super`. Solved

Thanks! This solution worked perfectly for me!

sarahalipour, 2025-04-07:
> Yep, in `($CONDA_PREFIX)/lib/python3.xx/site-packages/adafruit_platformdetect/constants/board.py`, find ORIN_NANO, then pick the board that you have (devkit: `nvidia,p3768-0000+p3767-0005`) and add a new line with `nvidia,p3768-0000+p3767-0005-super`. Solved

The error has been resolved, but I still can’t move the servo motors. Here’s what I did:

1. Ran the below command  in terminal
` python3 -c "import adafruit_platformdetect; print(adafruit_platformdetect.__file__)"`
     
   to display the path used by adafruit_platformdetect.

2. Navigated to the “constants” subdirectory and opened the *boards.py* file. In my case, the full path is:  
 ` /home/sahand/.local/lib/python3.10/site-packages/adafruit_platformdetect/constants/boards.py`

3. Searched for `nvidia,p3768-0000+p3767-0005` in the file, then added `nvidia,p3768-0000+p3767-0005-super` immediately after it.

This change fixed the board’s NotImplementedError. However, when I try to move the servos using:

```
from adafruit_servokit import ServoKit

myKit = ServoKit(channels=16)
myKit.servo[0].angle = 90
```
no errors are raised, yet the servos still don’t respond.

For context, I’m using a PCA9685 board with the following connections:  
- VCC → 3.3 volts (Jetson Pin 1)  
- GND → Ground (Jetson Pin 6)  
- SDA → Jetson Pin 3  
- SCL → Jetson Pin 5