adafruit/Adafruit_Blinka #956: Error with Raspberry Pi 5, Adafruit Blinka and Adafruit Circuitpython ADS1x15 (RPi.GPIO incompatibility problem)

dvelaren, 2025-03-11:
### Board Name

Raspberry Pi 5

### Steps

1. SSH into the board
2. Install Adafruit Blinka following this [link](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi) 
3. Reboot and ssh again into the board
4. Change directory to the previous created directory
5. Activate venv
6. Install Adafruit ADS1x15
```sh
pip3 install adafruit-circuitpython-ads1x15
```
7. Test provided blinkatest.py:
```python
import board
import digitalio
import busio

print("Hello, blinka!")

# Try to create a Digital input
pin = digitalio.DigitalInOut(board.D4)
print("Digital IO ok!")

# Try to create an I2C device
i2c = busio.I2C(board.SCL, board.SDA)
print("I2C ok!")

# Try to create an SPI device
spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
print("SPI ok!")

print("done!")
```
8. Check error:
```sh
RuntimeError: Cannot determine SOC peripheral base address
```

### Description

Hello Adafruit support,

I would like to report a compatibility problem when using both Adafruit Blinka (installed through automated install steps on this [link](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi)) with ADS1x15 library. It seems the problem is related to the installation of RPi.GPIO pip package after installing the ADS1x15 library via `pip3 install adafruit-circuitpython-ads1x15`.

I have been reading and the Adafruit Blinka library must remove RPi.GPIO and use instead rpi-lgpio.

When I run the blinkatest.py provided in the Adafruit Blinka link and after installing the Adafruit ADS1x15 library

```python
import board
import digitalio
import busio

print("Hello, blinka!")

# Try to create a Digital input
pin = digitalio.DigitalInOut(board.D4)
print("Digital IO ok!")

# Try to create an I2C device
i2c = busio.I2C(board.SCL, board.SDA)
print("I2C ok!")

# Try to create an SPI device
spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
print("SPI ok!")

print("done!")
```

I get the following error
```sh
RuntimeError: Cannot determine SOC peripheral base address
```

Please help us fix this problem so we can both use digital input and outputs of the RPI5 with the ADS1x15 module.

Thanks in advance,
David

caternuson, 2025-03-11:
Repeat (or re-emergence?) of #907?

dvelaren, 2025-03-11:
> Repeat (or re-emergence?) of [#907](https://github.com/adafruit/Adafruit_Blinka/issues/907)?

It is related however the problem is the combination usage of the blinka library with other adafruit libraries such as Adafruit Circuitpython ADS1115. Problem persist even if the RPi.GPIO is uninstalled and then the rpi-lgpio is installed...

caternuson, 2025-03-12:
That's the same as what is described in that issue thread. There's even a post explaining how to resolve the issue:
https://github.com/adafruit/Adafruit_Blinka/issues/907#issuecomment-2605477551

Does that work to fix it on your set up as well?

makermelissa, 2025-03-14:
Fixed by #911.

closed 2025-03-14 by makermelissa