adafruit/Adafruit_Blinka #963: Board detected and not detected

MswCoppola, 2025-04-23:
### Board Name

Adafruit FT232H

### Steps

Having followed https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/overiew I attempted to run the following commands 

*********************
import time
import sys
import os
import usb.util

os.environ["BLINKA_FT232H"] = "1"

dev = usb.core.find(idVendor=0x0403, idProduct=0x6014)
print(dev, "\n device is found\n")
import board
*********************

This failed, as such I attempted to understand the point of failure by manually calling the functions:

*********************
from adafruit_blinka.agnostic import board_id, detector
from adafruit_platformdetect import Detector

detector_manual = Detector()
print("Board id, using detector from platformdetect: ", detector_manual.board.id)
print("Board id, using detector from agnostic: ", detector.board.id)
print("Board id, using board_id from agnostic: ", board_id)

# Manually call the correct board from the list
import adafruit_blinka.board.ftdi_ft232h as board
print(dir(board))
*********************

This however, succeeded. 

[image]

[image]

### Description

The error happens when I try to use or even detect the board using the "import board" function.
After doing a bit of a dive into the code I found that the function that fails is the "board_id" function from the "adafruit_blinka.agnostic", which is called by the "board" module.

[image]

This is an issue that has plagued multiple developers using "adafruit_blinka" for which solutions seem to perform incosistently. However what surprises me is that the function above just runs the "board_id" function found in "adafruit_blinka.agnostic":

[image]

Which clearly just uses the detector function also in the "adafruit_blinka.agnostic" file. However when I instead call this detector function directly using: 

[image]

I am able to find the board ID.
What is going wrong and how could I fix it?

ladyada, 2025-05-02:
what is the host computer

caternuson, 2025-06-05:
@jlsilicon Unless @MswCoppola is also using an Orange Pi 5 / Pro, please don't hijack this issue thread.

@MswCoppola In addition to knowing the host computer, it would also help to know which post install check does not work:
https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/troubleshooting

makermelissa, 2025-08-26:
Closing due to lack of response.

closed 2025-08-26 by makermelissa