adafruit/Adafruit_Blinka #964: suport pulseio.pulseOut on Pi Version of Blinka

jo1525085251, 2025-04-28:
**Re: suport pulseio.pulseOut on Pi Version of Blinka**

Trying to use a Adafruit High Power Infrared IR LED Emitter with a Pi Zero (installed Blinka as per instructions in April 2025 - not too sure of CircuitPython version). 

I noticed this on attempting to use pulseio.PulseOut:

`AttributeError: module 'pulseio' has no attribute 'PulseOut'`

I looked at "adafruit_blinka.microcontroller.bcm283x.pulseio" and realised I can only find PulseIn as relevant code.

Is there a way to implement pulseio.pulseOut please for this platform?

ladyada, 2025-04-29:
not really, the pi is not able to generate 38khz PWM toggled signals. you need an arduino/microcontroller for that

jo1525085251, 2025-04-30:
Hi
Thanks for the prompt reply.
Actually I resolved my problem (generating infrared control codes for use with the Adafruit High Power Infrared IR LED Emitter) by using the pigpio library. Appears to work on initial testing with an infrared controlled desktop fan. 
I can go into more detail if you are interested.