adafruit/Adafruit_Blinka #966: lgpio_pin.py fails looking for driver after pi 5 debian update

pwagsc, 2025-05-01:
### Board Name

PI 5

### Steps

This was posted in PI forum:  My failure was identical and occurred after upgrade to 6.12.20+rpt-rpi-2712 on my PI 5. Peering at lgpio_pin.py code reveals that it fails when it tries to read the symbolic linked compatible file for gpiochip14 since the symlink does not exist. In fact gpiochip14 didn't exist before update. So the failure is the absent symlink and compatible file. All other gpiochipxx directories have the symlink and file. It is above my pay grade to determine the flaw that allowed creation of a new gpiochip that has no symlink. Some debian expert will have to sort that out. If it is legal to have no symlink then adafruit needs to fix their code (or for robustness). A simple try ... except pass around lines 18-26 and all is back to working again as it then ignores any gpiochip that doesn't have a compatible file/symlink. As a note, the .glob of files to be inspected just happened to start with gpiochip14 so it couldn't find chip0.
As a side note, I see the list of drivers doesn't include raspberrypi,bcm2712-gpio. However on my system driver 0 has raspberrypi,rp1-gpio in compatible file. So don't know what is going on there but it doesn't hurt anything at present.

### Description

Only happens after upgrade to 6.12.20+rpt-rpi-2712 on my PI 5

makermelissa, 2025-05-05:
Thanks, I'll test it out and see if I can reproduce the error.

makermelissa, 2025-05-05:
For reference, it appears your post is here: https://forums.raspberrypi.com/viewtopic.php?p=2313555&hilit=gpiochip14#p2312930

Do you have some test code you were running.

makermelissa, 2025-05-06:
Tested this out with 6.12.25+rpt-rpi-2712 (current latest) and did not run into these issues. Perhaps it was a bug in 6.12.20+rpt-rpi-2712 and they quickly released it again. Please update the Pi and give it another try. Also, to note, I think you meant `gpiochip4` rather than `gpiochip14`.

pwagsc, 2025-05-06:
Hi Melissa,

Am in the middle of an intense debug session.  Will follow up with you in my morning - US east coast.  For now, I want to clarify the issue as I understand it.  Adafruit blinka library is failing because there exists a gpiochip14 (not 4) that does not contain an at-nodesymbolic link which lgpio expects is there for every gpiochipxx so it can reference a compatible file.  So the bug is simple, either Adafruit should ignore checking gpiochipxx that have no symbolic link  OR whatever process is creating the gpiochipxx is failing to instantiate an at-node symbolic link.  Simple.

On my system I have a gpiochip0, gpiochip10, gpiochip11, gpiochip12, and gpiochip13 that all have a symbolic link to a compatible file.  However, gpiochip14 has no at-node link so blinka fails in lgpio when scanning for a compatible driver.  If on your test system all gpiochipxx have the symbolic link then it will not fail.

So why does my system have one without and yours doesn't?

In the AM I will save my debug env (for another problem) and reboot without my usb ssd.  I realized after seeing the latest post on the issue in pi forum that another person has the problem only after adding a usb device.  So if removing my usb solves the issue, then we know what caused the odd gpiochip14 (btw not 4 but 14) with no symlink.

If no fix I will update the OS to see if it goes away.

Then we will have narrowed the scope hopefully.  Will reply to you after test in AM.

Cheers,
Paul

________________________________
From: Melissa LeBlanc-Williams
Sent: Monday, May 5, 2025 8:51 PM
To: adafruit/Adafruit_Blinka
Cc: pwagsc; Author
Subject: Re: [adafruit/Adafruit_Blinka] lgpio_pin.py fails looking for driver after pi 5 debian update (Issue #966)

[https://avatars.githubusercontent.com/u/2681417?s=20&v=4]makermelissa left a comment (adafruit/Adafruit_Blinka#966)<https://github.com/adafruit/Adafruit_Blinka/issues/966#issuecomment-2852864573>

Tested this out with 6.12.25+rpt-rpi-2712 (current latest) and did not run into these issues. Perhaps it was a bug in 6.12.20+rpt-rpi-2712 and they quickly released it again. Please update the Pi and give it another try. Also, to note, I think you meant gpiochip4 rather than gpiochip14.

makermelissa, 2025-05-06:
Here's my gpiochip files on a fresh install (after updating):
```
(env) pi@raspberrypi:~ $ ls /dev/gpiochip*
/dev/gpiochip0  /dev/gpiochip10  /dev/gpiochip11  /dev/gpiochip12  /dev/gpiochip13  /dev/gpiochip4
(env) pi@raspberrypi:~ $ ls -l /dev/gpiochip*
crw-rw---- 1 root gpio 254,  0 May  5 17:41 /dev/gpiochip0
crw-rw---- 1 root gpio 254, 10 May  5 17:41 /dev/gpiochip10
crw-rw---- 1 root gpio 254, 11 May  5 17:41 /dev/gpiochip11
crw-rw---- 1 root gpio 254, 12 May  5 17:41 /dev/gpiochip12
crw-rw---- 1 root gpio 254, 13 May  5 17:41 /dev/gpiochip13
lrwxrwxrwx 1 root root       9 May  5 17:41 /dev/gpiochip4 -> gpiochip0
```

makermelissa, 2025-05-11:
Thanks, that give me more to test with.

makermelissa, 2025-05-20:
Closing for now because latest release includes fix. If you are still experiencing problems after updating to the latest release, feel free to reopen the issue.

closed 2025-05-20 by makermelissa