adafruit/Adafruit_Blinka #980: Supporting CS0 (aka D3) in FT232H board(s) for SPI protocol

cpg, 2025-06-21:
I'd love it if Blinka supported the "proper" chip select pin CS0 (aka D3) in the FT232H devices when using SPI.

The [details](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/pinouts) of this [Adafruit breakout board](https://learn.adafruit.com/adafruit-ft232h-breakout/overview) indicates that

    "CS0 - Chip Select is on D3. This is not used by Blinka, instead use one of the GPIO pins."

I was able to use a GPIO (D4), however, when using other programs in my hardware setup (e.g. the popular [flashrom](https://flashrom.org/) program), they use the default (D3) and only allow to configure the CS as D0..D3, which is a problem: I have to change hardware (use pin D3 or D4) and that's not doable as my device will be hardwired.

Looking through the code I do not find a clear reason why this is not possible. I wish I could test locally easily, as I am using VScode with and extension and I am not sure how to hack into this. 

It does appear that CS or CS0 is not defined as a pin in [ftdi_mpsse/ft232h/pin.py](src/adafruit_blinka/microcontroller/ftdi_mpsse/ft232h/pin.py). Is there a reason why or was it an oversight?

Can support for using the default be added easily? I could perhaps attempt it with some guidance.

caternuson, 2025-07-02:
This is a general requirement for Blinka. Blinka will manage the CS pin as part of its SPI support. Any hardware provided CS management is ignored.

Why do you need this capability in the context of using Blinka?

caternuson, 2025-07-02:
This seems like the general issue:

> I have to change hardware (use pin D3 or D4) and that's not doable as my device will be hardwired.

cpg, 2025-07-04:
Unfortunately, the other programs and libraries and the consisten use of D3 as CS0 sets me out from using Blinka.

Thanks for your answer that this is a Blinka thing.

closed 2025-07-04 by cpg