adafruit/Adafruit_Blinka #992: Support for Seeed Studio Xiao ESP32C6

pranavveenam, 2025-07-15:
Trying to use the Xiao ESP32C6 with the CircuitPython Extension for VSCode. I flashed the mcu with the latest stable release of the firmware found [here](https://circuitpython.org/board/seeed_xiao_esp32c6/), but when I try to run my program, i still get the NotImplementedError. I've selected the correct board from the I've run `pip3 install --upgrade adafruit-blinka adafruit-platformdetect` and see all the "Requirement already satisfied" messages.

Not sure what I'm doing wrong. Does running the program upload it to the board as well or is there a separate process to upload the code.py to the board?

For starters, I'm literally trying to recreate the Arduino "Blink" example in CircuitPython.

[image]

pranavveenam, 2025-07-16:
Okay, so after some digging, I've found that I am essentially unable to write to the board from the circuitpython extension. I can write a program in Thonny and see it work on the hardware and in the serial monitor in VSCode. the REPL in VSCode is also working. But I can't write to the board. Gemini says that the designed way to write files to the board to just save the file (ctrl+s) but that isn't doing anything. The serial monitor continues to show the the code I edited in Thonny

makermelissa, 2025-07-16:
This board is not supported by Blinka and likely won't ever be. Blinka is intended as a compatibility layer for Single Board Computers and a small handful of MicroPython boards. You are probably want CircuitPython instead. See https://circuitpython.org/board/seeed_xiao_esp32c6/.

closed 2025-07-16 by makermelissa