adafruit/Adafruit_Blinka #1007: Add new boards into a discussion

makermelissa, 2025-09-03:
In order to reduce the issues from being hidden amongst new board requests, they should be moved into a discussion. We will also need to put in place a way to direct users to add to the discussion rather than opening an issue.

brentru, 2025-10-24:
@makermelissa  Yeah, agreed. You may want to check out how the ESPHome project does it, I like their implementation when you click New Issue.

https://github.com/esphome/esphome/issues

makermelissa, 2025-10-25:
> [@makermelissa](https://github.com/makermelissa) Yeah, agreed. You may want to check out how the ESPHome project does it, I like their implementation when you click New Issue.
> 
> https://github.com/esphome/esphome/issues

Looks similar to the way we currently do it, except we have emojis 😺 .

closed 2026-04-07 by makermelissa